nicokaiser/rpi-audio-receiver #2: no bt device shown on iphone

Goeste, 2018-04-21:
Hi, I wanted to just use the bluetooth script, but no device shows up on my iphone when trying to find it. I can conect via ssh and bluetoothctl, but this should be prevented by udev i guess...

best regards,
goeste

nicokaiser, 2019-05-29:
Is this still an issue? Much has been improved in Raspbian during the last year, so probably this is fixed?

sumpfi82, 2019-07-01:
Yes it is still an issue! :(

nicokaiser, 2019-07-02:
Could you check the output of `sudo hciconfig`? It should say `UP RUNNING PSCAN ISCAN` somewhere. If not, you might need to execute

```
hciconfig hci0 piscan
hciconfig hci0 sspmode 1
`
``

to set the Bluetooth system in the right state.

sumpfi82, 2019-07-02:
Cool! It works! :)

nicokaiser, 2019-07-02:
Great! Can you try to restart you Pi to see if this is persistent? It should be (unless you modified the filesystem to be read-only, in which case you might need to make it writable again, restart the `bluetooth` service, and do these steps), but I'm not sure...

closed 2019-07-02 by nicokaiser

sumpfi82, 2019-07-02:
Yes, it seems to work even after reboot...