nicokaiser/rpi-audio-receiver #5: Directories don't exist?

jbonnett92, 2018-06-20:
On running the command `sudo ./install-bluetooth.sh` I got the following errors:
`./install-bluetooth.sh: 22: ./install-bluetooth.sh: cannot create /opt/local/bin/bluetooth-agent: Directory nonexistent`
`chmod: cannot access '/opt/local/bin/bluetooth-agent': No such file or directory`
`./install-bluetooth.sh: 174: ./install-bluetooth.sh: cannot create /opt/local/bin/bluetooth-udev: Directory nonexistent`
`chmod: cannot access '/opt/local/bin/bluetooth-udev': No such file or directory`

Any ideas?

Thanks,
Jamie

rhealey-uk, 2018-07-04:
I also had this issue. Installing on Pi Zero W with clean install of Raspbian Stretch Lite.

Until anyone has a more elegant solution I got around it by manually creating the directory ‘/opt/local/bin/‘ before running the install-Bluetooth.sh
This allowed the script to run with no errors.

Rick.

jbonnett92, 2018-07-09:
Ok after a fresh install and running `sudo ./install-all.sh` i get a lot of errors like this:
```
Err:3 http://www.mirrorservice.org/sites/archive.raspbian.org/raspbian stretch/main armhf fontconfig-config all 2.11.0-6.7
  404  Not Found
Err:11 http://www.mirrorservice.org/sites/archive.raspbian.org/raspbian stretch/main armhf libgstreamer1.0-0 armhf 1.10.4-1
  404  Not Found
Err:15 http://www.mirrorservice.org/sites/archive.raspbian.org/raspbian stretch/main armhf libtag1v5-vanilla armhf 1.11.1+dfsg.1-0.1
  404  Not Found
E: Failed to fetch http://www.mirrorservice.org/sites/archive.raspbian.org/raspbian/pool/main/c/cdparanoia/libcdparanoia0_3.10.2+debian-11_armhf.deb  404  Not Found
E: Failed to fetch http://www.mirrorservice.org/sites/archive.raspbian.org/raspbian/pool/main/f/fontconfig/fontconfig-config_2.11.0-6.7_all.deb  404  Not Found
E: Failed to fetch http://www.mirrorservice.org/sites/archive.raspbian.org/raspbian/pool/main/g/gstreamer1.0/libgstreamer1.0-0_1.10.4-1_armhf.deb  404  Not Found
E: Failed to fetch http://www.mirrorservice.org/sites/archive.raspbian.org/raspbian/pool/main/t/taglib/libtag1v5-vanilla_1.11.1+dfsg.1-0.1_armhf.deb  404  Not Found
```

nicokaiser, 2018-12-04:
I added an `mkdir -p /opt/local/bin` in `install-bluetooth.sh` in f2cc9dd5ab0cdfa51dc2029971cbe20c0cb1b2d3, so the directory should now be created automatically.

@jbonnett92 This looks like a network connectivity issue, can you check if your Raspi is connected correctly?

closed 2018-12-04 by nicokaiser