nicokaiser/rpi-audio-receiver #7: Bluetooth Auto-Connect

sumpfi82, 2018-09-17:
Hi! I'm really new on Raspberry but it works fine!
Is it possible to have a auto-connect my Smartphone via Bluetooth. Right now I have to connect the PI manually from my iPhone (6s) ... it's okay but it would be better if it connect when I'm in the near of my PI...

Thanks a lot!!!

nicokaiser, 2018-12-04:
I was thinking about this for a while. Some Bluetooth audio devices do this by remembering the last two paired devices and try to find them every now and them for auto connection.

However this can be very annoying when you not always want to connect your phone to the audio device, but only if you choose to. Permanent connections draw quite some battery.

I keep this open for further ideas, but have no plans to include this right now.

bchavet, 2019-10-13:
I solved this problem for my use case by adding an `ExecPostStart` step to the `bluetooth-agent` service. I use my rPi in my car to provide bluetooth audio over the AUX input, so I want it to try to reconnect to known clients on boot, when the car starts up.

First, a new script at `/usr/local/bin/bluetooth-reconnect.sh`, which tries to connect to known clients, starting with the most recently connected. It stops as soon as one connects.

```sh
#!/bin/bash

CONTROLLER=`ls /var/lib/bluetooth`
CLIENTS=`ls -t /var/lib/bluetooth/$CONTROLLER/cache`

for CLIENT in $CLIENTS; do
    if bluetoothctl connect $CLIENT; then break; fi
done
```

Then, I modified `/etc/systemd/system/bluetooth-agent.service`. I removed the `hciconfig` commands, because they never seemed to work on boot (making the service fail to start altogether), and don't appear to be necessary, but I don't really understand what they do so YMMV.

The autoconnect bit is the `ExecStartPost` line at the end

```
[Unit]
Description=Bluetooth Agent
Requires=bluetooth.service
After=bluetooth.target bluetooth.service

[Install]
WantedBy=multi-user.target

[Service]
Type=simple
#ExecStartPre=/bin/hciconfig hci0 piscan
#ExecStartPre=/bin/hciconfig hci0 sspmode 1
ExecStart=/usr/local/bin/bluetooth-agent
ExecStartPost=/usr/local/bin/bluetooth-reconnect.sh
```

martenumberto, 2020-03-07:
Is this still working?
The File `/etc/systemd/system/bluetooth-agent.service `doesnt exist in my installation. 
> 
> 
> I solved this problem for my use case by adding an `ExecPostStart` step to the `bluetooth-agent` service. I use my rPi in my car to provide bluetooth audio over the AUX input, so I want it to try to reconnect to known clients on boot, when the car starts up.
> 
> First, a new script at `/usr/local/bin/bluetooth-reconnect.sh`, which tries to connect to known clients, starting with the most recently connected. It stops as soon as one connects.
> 
> ```shell
> #!/bin/bash
> 
> CONTROLLER=`ls /var/lib/bluetooth`
> CLIENTS=`ls -t /var/lib/bluetooth/$CONTROLLER/cache`
> 
> for CLIENT in $CLIENTS; do
>     if bluetoothctl connect $CLIENT; then break; fi
> done
> ```
> 
> Then, I modified `/etc/systemd/system/bluetooth-agent.service`. I removed the `hciconfig` commands, because they never seemed to work on boot (making the service fail to start altogether), and don't appear to be necessary, but I don't really understand what they do so YMMV.
> 
> The autoconnect bit is the `ExecStartPost` line at the end
> 
> ```
> [Unit]
> Description=Bluetooth Agent
> Requires=bluetooth.service
> After=bluetooth.target bluetooth.service
> 
> [Install]
> WantedBy=multi-user.target
> 
> [Service]
> Type=simple
> #ExecStartPre=/bin/hciconfig hci0 piscan
> #ExecStartPre=/bin/hciconfig hci0 sspmode 1
> ExecStart=/usr/local/bin/bluetooth-agent
> ExecStartPost=/usr/local/bin/bluetooth-reconnect.sh
> ```

bchavet, 2020-03-09:
It's still working for me, but I also haven't touched my pi since I set it up. Looks like the `bluetooth-agent` service was renamed to `a2dp-agent` in January. https://github.com/nicokaiser/rpi-audio-receiver/commit/3d63b15de052695cbb54ca08f9a8168c2841cc49

nicokaiser, 2020-03-19:
Yes, I renamed the `bluetooth-agent` to `a2dp-agent` to match HiFiBerry's implementation on this. Sorry...

closed 2020-03-19 by nicokaiser