nicokaiser/rpi-audio-receiver #9: Feature: Sync multiple Pi's

jbonnett92, 2018-12-29:
Hi,
Is it possible we can a new feature to this? I was thinking something like Sonos where you can have multiple Pi's and sync the sound to each.

Jamie

nicokaiser, 2019-05-27:
Unfortunately, AirPlay 2 (which does support proper multi-speaker setups) is not (yet) reverse engineered, so currently the only way is to use iTunes and (multiple) AirPlay 1 (via shairport-sync) devices.

There is something called [Snapcast](badaix/snapcast), but I did not have the time to try it. Maybe integrating a Snapcast client (!) would be a no-brainer, but you would still need a server (and this project is about building the receiver-part of the setup).

jbonnett92, 2019-05-27:
@nicokaiser That link doesn't exist for me.

nicokaiser, 2019-05-27:
> @nicokaiser That link doesn't exist for me.

Sorry, the link was wrong, I fixed it in the comment.

closed 2019-10-22 by nicokaiser