nicokaiser/rpi-audio-receiver #12: don't use rpi-update

kle99, 2019-04-08:


kle99, 2019-04-08:
[Manual updates using rpi-update]([url](https://www.raspberrypi.org/documentation/linux/kernel/updating.md))

The rpi-update utility will download the latest (unstable, testing) kernel version and copy all required files onto your system. Note that the latest kernel from rpi-update is not guaranteed to work correctly!

kle99, 2019-04-08:
[Manual updates using rpi-update](https://www.raspberrypi.org/documentation/linux/kernel/updating.md)

The rpi-update utility will download the latest (unstable, testing) kernel version and copy all required files onto your system. Note that the latest kernel from rpi-update is not guaranteed to work correctly!

nicokaiser, 2019-05-27:
Fixed in ed102437ac48f4b2211819f1909f09aa046bea6d. The current (stable) Raspbian kernel works fine with Bluetooth now, no need to use the (potentially) unstable one from `rpi-update`.

closed 2019-05-27 by nicokaiser