nicokaiser/rpi-audio-receiver #13: Test Raspbian Buster compatibility

nicokaiser, 2019-06-26:


ductsoup, 2019-09-10:
I think you were asking for comments, if not just ignore me. 

Using Pi 3 + pHat DAC + Buster to prototype and your scripts work really well. I only ran into a couple minor Bluetooth issues.

- #15, I'm sure there's a better way but I just created a startup.sh and added @reboot in crontab to call it.
- The whole Bluetooth trust thing is inconvenient for this. I want it to work like a Rocketfish and talk to strangers. There's a script/hack a the bottom of this [doc](https://docs.google.com/document/d/12cK4heNd7kY3jYZI_sW1AD407dn_ds-9zZdX4qs-TJg/edit) that helps with that. Probably not the best solution but it's a way to go.

Next step is to do it all over with a PiZeroW then box it up. My friend can't let go of his HiFi system so I figured the least I could do is drag him into the 21st century so he can pair it with iTunes.

Thanks for sharing your work.

Sennevds, 2019-09-10:
Did a new install and noticed the agent wasn't running! I'm on the latest buster(2019-07-10) and the hcitool change from /usr/bin/ to /bin/

nicokaiser, 2019-09-10:
@Sennevds Did you check if there is a symlink from `/bin` to `/usr/bin`? I did not have a look at a new Raspbian installation since 2019-06-30 but I'm pretty sure if they included `usrmerge` there are symlinks. So `/usr/bin/hcitool` should still work.

@ductsoup See #16. Please check if the device is still not discoverable at startup. Should be done automatically now.

I also do not really like the Bluetooth trust thing, I feel uncomfortable having a device that anyone (near it) can play audio without even touching it. But I did not find an elegant was to toggle discoverability *without buttons or fiddling via ssh*.

Sennevds, 2019-09-10:
@nicokaiser Indeed there's a symlink. Weird because I got an error in the systemctl script it could not find the hcitool. When I changed the script to /bin/hcitool it works

ductsoup, 2019-09-10:
Fresh PiZeroW + Pimoroni pHat DAC + Buster ... definitely better. Still needed a light touch of SSH intervention to become discoverable and pair (default-agent, discoverable on).

Sennevds, 2019-09-13:
I fixed the bluetooth-agent by adding a restart on failure in the service.
The bluetooth.service takes sometime to start and the agent doesn't wait apparently.

cklam2, 2019-09-25:
I had the same problems on Buster. Took some time to figure out what to change. So here's what you have to do:
```sudo nano /etc/systemd/system/bluetooth-agent.service```
Under [Service] change ```/usr/bin/hciconfig``` to ```/bin/hciconfig```

Also under [Service] add the following 2 lines for restart on failure:
```Restart=always```
```RestartSec=10```

nvalis, 2019-11-04:
@Sennevds Does #26 fix this issue? Because I had the same problem and this was the solution I came up with.

closed 2019-11-05 by nicokaiser