nicokaiser/rpi-audio-receiver #14: Getting HifiBerry to work!?

sumpfi82, 2019-07-02:
Everything looks fine after the installation. Airplay and Bluetooth is getting connected, but there is NO SOUND coming out...
I have a Pi 3 with HifiBerry MiniAmp
I tried the enable-hifiberry.sh, but it says I have no permission. I manually switched dtoverlay to hifiberry-dac in /boot/config.txt but it doesn't work. :(

nicokaiser, 2019-07-15:
You need to execute the scripts with "sudo", e.g. `sudo ./enable-hifiberry.sh`. Could you try this?
I might need to adjust the scripts so they do the "sudo" part by themselves.

pendrya, 2019-08-07:
May I slightly derail this thread, as I think this will be helpful for others. I'm trying to add equalizer into mix with Hifiberry DAC+, however after hours of googling/reading the ALSA docs I still cannot get the right `asound.conf` configuration. I can't figure out the correct config to make following source device to rpi-audio-receiver (either via bt or airplay) to hifiberry amp2 via equalizer. Could anyone share their knowledge wether this possible?

nicokaiser, 2019-08-08:
What is your current asound.conf? And which equalizer are you using, alsaequal?

pendrya, 2019-08-08:
> What is your current asound.conf? And which equalizer are you using, alsaequal?

My current asound.conf looks like this:
`pcm.dmixer {
  type dmix
  ipc_key 1024
  ipc_perm 0666
    slave {
    pcm "hw:1,0"
    period_time 0
    period_size 1024
    buffer_size 8192
    rate 44100
  }
  bindings {
    0 0
    1 1
  }
}
ctl.dmixer {
  type hw
  card 0
}
pcm.softvol {
  type softvol
  slave.pcm "dmixer"
  control {
    name "Master"
    card 0
  }
}
pcm.!default {
  type plug
  slave.pcm "softvol"
}
`
With this setup I have sound coming out of HifiBerry DAC+. As soon as I add alsaequal either before the softvol or after I have nothing out of the speakers via rpi-audio-receiver, but if I test manually with player specifying the output device I can make it play and equalizer works.

nicokaiser, 2019-08-08:
So you "only" need to figure out how to configure the defaults so they work without having to specify the output device. How does your config look like with alsaequal added and how do you call aplay so it works with the equalizer?

pendrya, 2019-08-09:
> So you "only" need to figure out how to configure the defaults so they work without having to specify the output device. How does your config look like with alsaequal added and how do you call aplay so it works with the equalizer?

The asound.conf with eq looks like this:
`pcm.dmixer {
  type dmix
  ipc_key 1024 # any unique value
  ipc_perm 0666 # mixing for all users
  slave {
    pcm "hw:1,0"
    channels 2
    period_time 0
    period_size 1024
    rate 44100
    buffer_size 16384
  }
  bindings {
    0 1
    1 0
  }
}
pcm.softvol {
  type softvol
  slave.pcm "dmixer"
  control {
    name "Master"
    card 0
  }
}
ctl.equal {
  type equal
}
pcm.plugequal {
  type equal
  slave.pcm plug:softvol
}
pcm.equal {
  type plug;
  slave.pcm plugequal
}
pcm.!default {
  type plug
  slave.pcm plugequal
}
`
I'm testing it with speaker-test command and mpg123 player and aplay command: `speaker-test -D default -c 2` & `mpg123 test.mp3` & `aplay test.wav` — all three commands successfully play sound, equalizer works, softvol 'Master' volume works, however when I switch to airplay/bluetooth — no sound.

nicokaiser, 2019-08-10:
That's strange. I did not test your config yet, but it seems okay. I assume you already restarted the services (shairport-sync, bluealsa)?

I had similar problems with pivumeter, a graphical VU meter using LEDs. It seems that dmixer does not play well with meter plugins, and I could not find any documentation about it, so I gave up. In your case it seems different because `aplay` actually works, so it seems to me only a small part is missing (i.e. make the services do the same thing aplay does).

pendrya, 2019-08-13:
> That's strange. I did not test your config yet, but it seems okay. I assume you already restarted the services (shairport-sync, bluealsa)?
>
> I had similar problems with pivumeter, a graphical VU [...]

UPDATE: 
Well I've managed to make it work over bluetooth, the thing was that when it plays through bluealsa it uses root as a user (I guess?) therefore alsamixer should be launched from root in order to adjust eq. I'm still digging to airplay config, as It seems it uses different user for playback

HorstBaerbel, 2020-03-16:
Try to use "ipc_key_add_uid false" in your dmixer config. That said, I removed my custom dmixer config, as it should be automagically created.
I use a "Adafruit speaker bonnet" and my working asound.conf (local audio + bluetooth) looks like this:
```
# Use soft volume control because or card does not have hardware volume control
pcm.!default {
  type plug
  slave.pcm "softvol" # redirect output to softvol (instead of "hw:0,0")
}

# Output to dmixer first. This enables:
# - automatic conversion to a format that the hardware "hw:0,0" can play
# - playing multiple simultaneous streams
pcm.softvol {
  type softvol
  slave {
    pcm "dmix" # redirect the output to dmix (instead of "hw:0,0")
}
  control {
    name "PCM" # override the volume slider to set the softvol volume
    card 0
  }
}
```

closed 2020-04-17 by nicokaiser