nicokaiser/rpi-audio-receiver #19: Stuttering problem

amilino, 2019-08-09:
Heavy stuttering every few seconds. I disabled WiFi in /boot/config.txt and I am on RPi4 Buster. I only installed install-bluetooth.sh.

nicokaiser, 2019-09-10:
Do you use an external Bluetooth adapter? The internal one of the Raspberry Pi is not really capable of stutter-free audio (even the Pi 4 seemingly...)

amilino, 2019-09-10:
No, I used integrated, but on RPi3 it worked well.

nicokaiser, 2019-09-10:
Okay, so it seems like the new Bluetooth module (or something else) is broken or not supported, again.

Did you test it with the sending device near the Raspberry? For the older devices I experienced heavy issues when not exactly near the device (i.e. the antenna is bad).

danielgusvt, 2019-10-14:
I think I can confirm this issue, this happens on my rpi 3 running the latest raspbian buster. I have to say that I also disabled wifi on /boot/config.txt and my bluetooth device is less than a metre near the raspberry and I get a slight stuttering which it is quite annoying. I don't think it is a hardware issue since it used to work fine, most likely it is a software issue from upstream (not this tool) so I will try to downgrade things here and there and I will report back soon.

closed 2019-11-05 by nicokaiser

kiai72, 2020-01-12:
since i also have stuttering (3b+) - i tried to disable wlan by 'sudo ifconfig wlan0 down'.
But this does not help.
How exactly do you disable WLAN in /boot/config.txt - maybe i shoudl try this.
Or what other options are there ?

nicokaiser, 2020-01-12:
If disabling WLAN with `ifconfig` does not help,  neither will disabling it in `config.txt`. The best in this case is to buy a cheap Bluetooth 4.0 dongle and disable the internal Bluetooth (`dtoverlay=pi3-disable-bt` in  `config.txt`). The Raspberry Pi 3 Bluetooth chip is okay for input devices, but not for audio...