nicokaiser/rpi-audio-receiver #21: First time pairing question.

belgarathcjd, 2019-09-18:
Hello, 
I'm not sure if what I am experiencing is a bug or this is how you designed this to work. The first time I attempt to pair a device with my pi2B (using a Bluetooth dongle) I am unable to pair the devices without using bluetoothctl. Once that initial pairing is complete everything runs brilliantly. Is my install running as designed or has something gone wrong? I'd love if I didn't have to do anything whenever a new device pairs.

johann-lr, 2019-12-01:
nearly the same issue... I've to validate a pairing key on my phone when I want to connect my device

silverbullettruck2001, 2020-09-01:
@nicokaiser I have been trying to use this on my Rpi3 without success.  I have disabled wifi on the Rpi3 to account for the issues with both wifi and bt being enabled, but even with that I cannot get my device to pair.  The device I am using is a TaoTronics TT-BA09 Transmitter.  Here is a link to the manual. [TT-BA09 Manual](https://cdn.shopify.com/s/files/1/0273/0351/0069/files/TT-BA09-User_Manual.pdf?3809).  I am using the transmitter in TX mode as I want to connect it to my non-bluetooth TV and stream audio from the TV to the RPi3.  I put the device in pairing mode and it never pairs.  If I try to connect a pair of bluetooth headphones to the TT-BA09, it works as would be expected and the audio plays.  Do you have any suggestions on how I can get this device to pair to the RPi3?  Let me know if there's anything I can provide to assist in troubleshooting this with me.  Thanks!

nicokaiser, 2020-10-31:
Can you try again? I changed the pairing mechanism to use the default `bthelper` script instead of a custom Python script. (a1e6ebd7d6055d002a215de9ae5f306dbf0b8ef7)

akamming, 2021-01-16:
i have the same issue. On my phone i could not see the device. I worked around it by manually giving the command "bluetoothctl discoverable on" after which the device could be found and everything worked fine

nicokaiser, 2021-01-16:
@akamming Can you post the output of "sudo hciconfig" before running `bluetoothctl discoverable on`? And  `sudo grep bluetoothctl /var/log/daemon.log`. The startup script I added in october should enable discoverability...

akamming, 2021-01-16:
ofcourse

sudo hiconfig:

```
hci0:   Type: Primary  Bus: UART
        BD Address: DC:A6:32:F4:49:7C  ACL MTU: 1021:8  SCO MTU: 64:1
        UP RUNNING PSCAN ISCAN
        RX bytes:1591 acl:0 sco:0 events:101 errors:0
        TX bytes:2365 acl:0 sco:0 commands:101 errors:0

```

and grep bluetootctl /var/log/daemon.log
```
Jan 16 12:05:23 Keuken bluetoothctl[417]: Changing discoverable on succeeded
Jan 16 12:05:23 Keuken bluetoothctl[564]: Changing discoverable on succeeded
Jan 16 12:07:33 Keuken bluetoothctl[563]: Changing discoverable on succeeded
Jan 16 12:38:14 Keuken bluetoothctl[561]: Changing discoverable on succeeded

```
(12:38:14 logline was after a reboot,  so no manual bluetoothctl command given yet)

bluetoothctl show gives me
```
Controller DC:A6:32:F4:49:7C (public)
        Name: Keuken
        Alias: Keuken
        Class: 0x00040414
        Powered: yes
        Discoverable: yes
        Pairable: yes
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v1D6Bp0246d0532
        Discovering: no
```
which looks good, but the device "keuken" is not discoverable  until i give the "bluetoothctl discoverable on" command.

Just found out it gets even weirder: "sudo bluetoothctl discoverable on" does not make any difference, but a normal "bluetoothctl discoverable on" (so it run as pi user)  really switches on discoverability.

This is a fresh raspberry image btw (installed it yesterday with latest raspbian image from pi website, sudo apt update && sudo apt upgrade -y and then using your scripts to make it an audio receiver).

belgarathcjd, 2021-01-18:
I finally got around to trying this again. I wiped the sd card completely and tried a fresh install. Everything worked perfectly first try. 
Thank you man, keep up the good work!

closed 2021-01-18 by belgarathcjd