nicokaiser/rpi-audio-receiver #31: mono to stereo

valix85, 2019-12-31:
Hi, when i run $ amixer I obtain this:
`Simple mixer control 'PCM',0                                                                                                                                                   Capabilities: pvolume pvolume-joined pswitch pswitch-joined                                                                                                                  Playback channels: Mono                                                                                                                                                      Limits: Playback -10239 - 400                                                                                                                                                Mono: Playback -25 [96%] [-0.25dB] [on] `

it's possible evolve it from mono to stereo?
Another question, sometimes audio pick/sick. I use raspberry 3 model B+ and I have connect it to yamaha rx-v571 by hdmi and I stream with OnePlus 7 by bluetooth.

Thank you

nicokaiser, 2020-01-08:
Can you test if the audio is really mono? As far as I know the output you mention only applies to the mixer (so both channels are adjusted equally when setting the volume). Do you try to play audio through the Raspberry's 3.5mm jack, or through HDMI?

nicokaiser, 2020-01-08:
As for the audio glitches: do you use the internal Bluetooth _and_ the internal WiFi? If so, try disabling the WiFi connection while playing via Bluetooth by uncommenting the two `ifconfig` lines in `/usr/local/bin/bluetooth-udev`.

To achieve the best result, you might need to use a Bluetooth USB dongle (like the CSR 4.0 / CSR8510) and a USB or I2S audio adapter.

valix85, 2020-01-10:
I test my system with youtube video ( https://www.youtube.com/watch?v=MkObqRUvpRY ) and work in stereo mode and work fine. I use HDMI under bluetooth.

valix85, 2020-01-10:
I try with 5.1 test ( https://www2.iis.fraunhofer.de/AAC/multichannel.html first HE-AAC ) but doesn't work fine, I think the sound in downgraded to Stereo.

nicokaiser, 2020-01-10:
Oh, I‘m not sure if BlueAlsa (or even Bluetooth Audio with SBC at all) supports more channels than Stereo...

closed 2020-04-17 by nicokaiser