nicokaiser/rpi-audio-receiver #32: doesn't work with google home (mini)

kiai72, 2020-01-05:
Hello,

everything works fine, when i connect a android phone to this receiver.
But i can't connect my google home to it - at least not when i use the "official way" by initiating the connection via google home app:

that's what i find in the console:

[NEW] Device 20:DF:B9:93:2E:9B Wohnzimmer
[CHG] Device 20:DF:B9:93:2E:9B Modalias: bluetooth:v000Fp1200d1436
[CHG] Device 20:DF:B9:93:2E:9B UUIDs: 0000110a-0000-1000-8000-00805f9b34fb
[CHG] Device 20:DF:B9:93:2E:9B UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Device 20:DF:B9:93:2E:9B UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Device 20:DF:B9:93:2E:9B UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Device 20:DF:B9:93:2E:9B UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Device 20:DF:B9:93:2E:9B UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Device 20:DF:B9:93:2E:9B UUIDs: 0000fea0-0000-1000-8000-00805f9b34fb
[CHG] Device 20:DF:B9:93:2E:9B ServicesResolved: yes
[CHG] Device 20:DF:B9:93:2E:9B Paired: yes
[CHG] Device 20:DF:B9:93:2E:9B ServicesResolved: no
[CHG] Device 20:DF:B9:93:2E:9B Connected: no

But then i've figured out: when i initiate the connection from the RPI -> home it works:
sudo bluetoothctl
trust 20:DF:B9:93:2E:9B
pair 20:DF:B9:93:2E:9B
connect 20:DF:B9:93:2E:9B

[bluetooth]# connect 20:DF:B9:93:2E:9B
Attempting to connect to 20:DF:B9:93:2E:9B
[CHG] Device 20:DF:B9:93:2E:9B Connected: yes
Connection successful
[CHG] Device 20:DF:B9:93:2E:9B ServicesResolved: yes
[CHG] Controller B8:27:EB:DC:A6:50 Discoverable: no

any idea ?

mwalasz, 2020-01-05:
I had exactly the same issue, but on Win 10. Spotify Connect, Airplay works like a charm, besides Bluetooth which needed registering device on Raspberry Pi 3+. Apart from this issue, really great job!

nicokaiser, 2020-01-08:
Can you have a look if there is something in the Raspberry's logs when trying to connect from Google Home? `journalctl -f` (and `logread -F` if you applied the read-only modifications) may be helpful.

kiai72, 2020-01-12:
strange. for some reason it worked now.
here's teh log:
Jan 12 13:20:47 raspberrypi bluetoothd[569]: Endpoint registered: sender=:1.5 path=/org/bluez/hci0/A2DP/SBC/Sink/3
Jan 12 13:20:48 raspberrypi bluetoothd[569]: /org/bluez/hci0/dev_20_DF_B9_93_2E_9B/fd0: fd(23) ready
Jan 12 13:20:49 raspberrypi kernel: input: 20:DF:B9:93:2E:9B as /devices/virtual/input/input9
Jan 12 13:20:50 raspberrypi systemd-udevd[822]: Process '/usr/sbin/th-cmd --socket /var/run/thd.socket --passfd --udev' failed with exit code 1.
Jan 12 13:20:50 raspberrypi systemd-logind[424]: Watching system buttons on /dev/input/event9 (20:DF:B9:93:2E:9B)

i keep watching this. in case it does not work, i'll report back.
thank you very much

nicokaiser, 2020-01-12:
Great! Let me know if you experience problems again, I do not have a Google Home...

closed 2020-01-12 by nicokaiser