nicokaiser/rpi-audio-receiver #33: Low Volume

valix85, 2020-01-10:
Hi, everytime system up I need to connect it via ssh and set volume with this command

`amixer sset 'PCM' 96%`

It's possible save this settings or automate it?

thank you

MattCosturos, 2020-01-13:
You could probably add that line to   /usr/local/bin/bluetooth-udev 
That script is called every time a device connects/disconnects.

docgloucester, 2020-02-01:
There's a way to save your alsamixer config.
When you're done tweaking your volume levels the way you want, just run 
`sudo alsactl store`
and they will be saved.

nicokaiser, 2020-02-01:
Thanks @docgloucester. Yes, `alsactl store` will store the current volume.

I did not add this to the installation script because some users may want to use an external device (USB or I2S) for much improved audio quality – and then they need to reboot the system before begin able to set the volume (And, the mixer may not be called "PCM" then, so setting the volume on startup would also not work).

closed 2020-04-17 by nicokaiser