nicokaiser/rpi-audio-receiver #36: Spotify Connect client drroping out and crashes

andlill, 2020-01-20:
Hi 
After a few songs played of a queue the connection to the client drops and crashes. 
A reboot is requiered to get things up and running again. 

Tried both Android phone and Windows PC Spotify client and samme result.

Any ideas of what the problem can be, is there any logs i can check for faults ? 

Best Regards 
Anders

nicokaiser, 2020-01-29:
Could you try to install the latest spotifyd version by using this script?

https://github.com/nicokaiser/rpi-audio-receiver/blob/9ecd480c8f5c178de53c17fde256b7f96f568445/install-spotify.sh

andlill, 2020-01-29:
Hi 
Can't get the connect client up and running when instaling from this new script. 

Seams like the installer do nothing even with a Y input on this package 

Please advice.

nicokaiser, 2020-02-01:
I'm sorry, I messed up the Spotifyd installation script, which did crash before copying the binary to its destination. Can you try again with the newest version?

nicokaiser, 2020-04-17:
This got better with spotifyd 0.2.24, however it may still be an upstream issue, see Spotifyd/spotifyd#458.

closed 2020-04-17 by nicokaiser