nicokaiser/rpi-audio-receiver #37: Bluetooth receiver not working

yanone, 2020-01-25:
Hi,
I've recently reinstalled the Pi after I messed up some configuration, and wanted to let you know that the thing didn't work anymore with your latest code. In particular, the Bluetooth wouldn't advertise itself and therefore can't be discovered.
I checked out [this commit](https://github.com/nicokaiser/rpi-audio-receiver/commit/7e9cab66c4480c337f9db883eddeeb8009172c7b) directly after my PR and got it working instantly.
Somewhere after that point something went wrong. This is on a Pi3.

projectdelphai, 2020-01-27:
I wonder if it's the fact that the script at /etc/systemd/system/btinerface.service references /usr/bin/hciconfig but hciconfig is really at /bin/hciconfig. I noticed errors in journalctl and after changing the systemd file to /bin/hciconfig, everything works now

nicokaiser, 2020-01-29:
Can you try again? Sorry, i messed up my development machine by running `usrmerge` as I thought this would be default in Rasbpian 10...

docgloucester, 2020-01-31:
Hi !
I'm encountering the same issue with no bluetooth advertisement whatsoever on my RPi 3B+, even after your latest commits.

I'm getting errors concerning bluetooth in journalctl, here's it :

`janv. 31 17:28:02 raspberrypi hciconfig[530]: Can't get device info: No such device
janv. 31 17:28:02 raspberrypi systemd[1]: btinterface.service: Control process exited, code=exited, status=1/FAILURE
janv. 31 17:28:02 raspberrypi kernel: Bluetooth: Core ver 2.22
janv. 31 17:28:02 raspberrypi kernel: NET: Registered protocol family 31
janv. 31 17:28:02 raspberrypi kernel: Bluetooth: HCI device and connection manager initialized
janv. 31 17:28:02 raspberrypi kernel: Bluetooth: HCI socket layer initialized
janv. 31 17:28:02 raspberrypi kernel: Bluetooth: L2CAP socket layer initialized
janv. 31 17:28:02 raspberrypi kernel: Bluetooth: SCO socket layer initialized
janv. 31 17:28:02 raspberrypi systemd[1]: btinterface.service: Failed with result 'exit-code'.
janv. 31 17:28:02 raspberrypi systemd[1]: Failed to start Bluetooth hardware initialisation.
janv. 31 17:28:02 raspberrypi systemd[1]: Dependency failed for BlueALSA aplay.
janv. 31 17:28:02 raspberrypi systemd[1]: bluealsa-aplay.service: Job bluealsa-aplay.service/start failed with result 'dependency'.
janv. 31 17:28:02 raspberrypi systemd[1]: Started BluezALSA proxy.`

I'm running the latest stable Raspbian.

Thank you for your time !

nicokaiser, 2020-01-31:
Can you send me the output of „sudo journalctl“? If it’s possible, the whole log, you can create it e.g. with „sudo journalctl > file.txt“.

yanone, 2020-02-01:
This is the log directly after boot. I installed from the zip file you linked (Why not from the sources anymore?). Still no Bluetooth visible.

```
-- Logs begin at Sat 2020-02-01 10:02:47 GMT, end at Sat 2020-02-01 10:04:26 GMT. --
Feb 01 10:02:47 lotus kernel: Booting Linux on physical CPU 0x0
Feb 01 10:02:47 lotus kernel: Linux version 4.19.93-v7+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1290 SMP Fri Jan 10 16:39:50 GMT 2020
Feb 01 10:02:47 lotus kernel: CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
Feb 01 10:02:47 lotus kernel: CPU: div instructions available: patching division code
Feb 01 10:02:47 lotus kernel: CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Feb 01 10:02:47 lotus kernel: OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
Feb 01 10:02:47 lotus kernel: Memory policy: Data cache writealloc
Feb 01 10:02:47 lotus kernel: cma: Reserved 8 MiB at 0x3ac00000
Feb 01 10:02:47 lotus kernel: On node 0 totalpages: 242688
Feb 01 10:02:47 lotus kernel:   Normal zone: 2133 pages used for memmap
Feb 01 10:02:47 lotus kernel:   Normal zone: 0 pages reserved
Feb 01 10:02:47 lotus kernel:   Normal zone: 242688 pages, LIFO batch:63
Feb 01 10:02:47 lotus kernel: random: get_random_bytes called from start_kernel+0xac/0x4b4 with crng_init=0
Feb 01 10:02:47 lotus kernel: percpu: Embedded 16 pages/cpu s36864 r8192 d20480 u65536
Feb 01 10:02:47 lotus kernel: pcpu-alloc: s36864 r8192 d20480 u65536 alloc=16*4096
Feb 01 10:02:47 lotus kernel: pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
Feb 01 10:02:47 lotus kernel: Built 1 zonelists, mobility grouping on.  Total pages: 240555
Feb 01 10:02:47 lotus kernel: Kernel command line: coherent_pool=1M 8250.nr_uarts=0 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:EE:C3:6B vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=6c586e13-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
Feb 01 10:02:47 lotus kernel: Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
Feb 01 10:02:47 lotus kernel: Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
Feb 01 10:02:47 lotus kernel: Memory: 939084K/970752K available (8192K kernel code, 653K rwdata, 2220K rodata, 1024K init, 822K bss, 23476K reserved, 8192K cma-reserved)
Feb 01 10:02:47 lotus kernel: Virtual kernel memory layout:
                                  vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                                  fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                                  vmalloc : 0xbb800000 - 0xff800000   (1088 MB)
                                  lowmem  : 0x80000000 - 0xbb400000   ( 948 MB)
                                  modules : 0x7f000000 - 0x80000000   (  16 MB)
                                    .text : 0x(ptrval) - 0x(ptrval)   (9184 kB)
                                    .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
                                    .data : 0x(ptrval) - 0x(ptrval)   ( 654 kB)
                                     .bss : 0x(ptrval) - 0x(ptrval)   ( 823 kB)
Feb 01 10:02:47 lotus kernel: SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Feb 01 10:02:47 lotus kernel: ftrace: allocating 26853 entries in 79 pages
Feb 01 10:02:47 lotus kernel: rcu: Hierarchical RCU implementation.
Feb 01 10:02:47 lotus kernel: NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Feb 01 10:02:47 lotus kernel: arch_timer: cp15 timer(s) running at 19.20MHz (phys).
Feb 01 10:02:47 lotus kernel: clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
Feb 01 10:02:47 lotus kernel: sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
Feb 01 10:02:47 lotus kernel: Switching to timer-based delay loop, resolution 52ns
Feb 01 10:02:47 lotus kernel: Console: colour dummy device 80x30
Feb 01 10:02:47 lotus kernel: console [tty1] enabled
Feb 01 10:02:47 lotus kernel: Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
Feb 01 10:02:47 lotus kernel: pid_max: default: 32768 minimum: 301
Feb 01 10:02:47 lotus kernel: Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
Feb 01 10:02:47 lotus kernel: Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
Feb 01 10:02:47 lotus kernel: CPU: Testing write buffer coherency: ok
Feb 01 10:02:47 lotus kernel: CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
Feb 01 10:02:47 lotus kernel: Setting up static identity map for 0x100000 - 0x10003c
Feb 01 10:02:47 lotus kernel: rcu: Hierarchical SRCU implementation.
Feb 01 10:02:47 lotus kernel: smp: Bringing up secondary CPUs ...
Feb 01 10:02:47 lotus kernel: CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
Feb 01 10:02:47 lotus kernel: CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
Feb 01 10:02:47 lotus kernel: CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
Feb 01 10:02:47 lotus kernel: smp: Brought up 1 node, 4 CPUs
Feb 01 10:02:47 lotus kernel: SMP: Total of 4 processors activated (153.60 BogoMIPS).
Feb 01 10:02:47 lotus kernel: CPU: All CPU(s) started in HYP mode.
Feb 01 10:02:47 lotus kernel: CPU: Virtualization extensions available.
Feb 01 10:02:47 lotus kernel: devtmpfs: initialized
Feb 01 10:02:47 lotus kernel: VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[... cut ...]
Feb 01 10:03:30 lotus systemd-timesyncd[257]: Synchronized to time server for the first time [2a03:4000:31:461::bad:c0de]:123 (2.debian.pool.ntp.org).
Feb 01 10:03:35 lotus dhcpcd[533]: eth0: fe80::9a9b:cbff:fecc:540c is reachable again
Feb 01 10:03:35 lotus dhcpcd[533]: eth0: fe80::9a9b:cbff:fecc:540c is reachable again
Feb 01 10:03:35 lotus systemd[1]: systemd-hostnamed.service: Succeeded.
Feb 01 10:04:26 lotus sudo[631]:       pi : TTY=pts/0 ; PWD=/home/pi ; USER=root ; COMMAND=/bin/journalctl
Feb 01 10:04:26 lotus sudo[631]: pam_unix(sudo:session): session opened for user root by pi(uid=0)
```

docgloucester, 2020-02-01:
Here's the full log of my journalctl.

Thank you for your time :)
[journalctl.txt](https://github.com/nicokaiser/rpi-audio-receiver/files/4142524/journalctl.txt)

nicokaiser, 2020-02-01:
Can you change the line in /etc/systemd/system/btinterface.service

```
After=bluetooth.service
```

to

```
After=bluetooth.target
```

and see if this helps? I'll check later on a new Raspberry Pi 3, but maybe this helps already...

docgloucester, 2020-02-01:
I haven't noticed any difference...

Here's the log
[journalctl_log2.txt](https://github.com/nicokaiser/rpi-audio-receiver/files/4142554/journalctl_log2.txt)

yanone, 2020-02-01:
> I haven't noticed any difference...

Same here

nicokaiser, 2020-02-01:
Either systemd or the Raspberry Pi Bluetooth initialization seems to be a bit broken concerning dependencies. Can you please try the following:

Remove btinterface.service:

    systemctl disable /etc/systemd/system/btinterface.service
    rm /etc/systemd/system/btinterface.service

Remove the dependency of btinterface in /etc/systemd/system/bluealsa-aplay.service:

    cat <<'EOF' > /etc/systemd/system/bluealsa-aplay.service
    [Unit]
    Description=BlueALSA aplay
    Requires=bluealsa.service
    After=bluealsa.service sound.target

    [Service]
    Type=simple
    User=root
    ExecStartPre=/bin/sleep 2
    ExecStart=/usr/bin/bluealsa-aplay --pcm-buffer-time=250000 00:00:00:00:00:00
    RestartSec=5
    Restart=always

    [Install]
    WantedBy=multi-user.target
    EOF
    systemctl daemon-reload
    systemctl enable bluealsa-aplay

Move the device initialisation to `bthelper` by adding a new file:

    # Make Bluetooth discoverable after initialisation
    mkdir -p /etc/systemd/system/bthelper@.service.d
    cat <<'EOF' > /etc/systemd/system/bthelper@.service.d/override.conf
    [Service]
    ExecStartPost=/usr/bin/bluetoothctl discoverable on
    ExecStartPost=/bin/hciconfig %I piscan
    ExecStartPost=/bin/hciconfig %I sspmode 1
    EOF

This ensures that the discoverability is set after the device is initialized by `bthelper`, which hopefully works for the internal Broadcom Bluetooth stack, as well as with external USB dongles, I'll check this later.

docgloucester, 2020-02-01:
That seems to have solved it. It now advertises itself and I can connect to it and broadcast audio.

However, my bluetooth stream is very choppy and it seems a lot of frames are being lost. I suspect the issue is completely unrelated, though.

nicokaiser, 2020-02-01:
Thanks for testing! 

Yes, the choppy audio is because the Raspberry Pi Bluetooth chip is crap and interferes with Wi-Fi. You can try to un-comment the two `ifconfig` lines in `/usr/local/bin/bluetooth-udev`, this disables Wi-Fi while a Bluetooth devices is connected. But you may consider buying a cheap CSR 4.0 Bluetooth USB dongle and disable the Pi's Bluetooth with `dtoverlay=pi3-disable-bt` in `/boot/config.txt`.

commit 265fc082 (2020-01-29): fix: Correct paths for hciconfig (fixes #37)
commit 06bba3f9 (2020-02-01): fix: Improve Bluetooth startup again. Fixes #37
closed 2020-02-01 by nicokaiser