nicokaiser/rpi-audio-receiver #38: Choppy bluetooth audio

docgloucester, 2020-02-01:
Hi,
I managed to make my Raspberry Pi 3B+ (running latest stable Raspbian) discoverable on Bluetooth and to have an audio stream to it.

However, the audio is very choppy and quite poor.
Moreover, I'm getting these lines in loop in journalctl when there's audio streamed to it.

```
févr. 01 18:18:40 raspberrypi kernel: Bluetooth: hci0: Frame reassembly failed (-84)
févr. 01 18:18:40 raspberrypi kernel: Bluetooth: hci0: Frame reassembly failed (-84)
févr. 01 18:18:40 raspberrypi kernel: Bluetooth: hci0: Frame reassembly failed (-84)
févr. 01 18:18:40 raspberrypi kernel: Bluetooth: hci0: Frame reassembly failed (-84)
févr. 01 18:18:40 raspberrypi kernel: Bluetooth: hci0: Frame reassembly failed (-84)
févr. 01 18:18:40 raspberrypi kernel: Bluetooth: hci0: Frame reassembly failed (-84)
févr. 01 18:18:40 raspberrypi kernel: Bluetooth: hci0: Frame reassembly failed (-84)
févr. 01 18:18:40 raspberrypi kernel: Bluetooth: hci0: Frame reassembly failed (-84)
févr. 01 18:18:40 raspberrypi kernel: Bluetooth: hci0: Frame reassembly failed (-84)
févr. 01 18:18:40 raspberrypi kernel: Bluetooth: hci0: Frame reassembly failed (-84)
févr. 01 18:18:40 raspberrypi kernel: Bluetooth: hci0: Frame reassembly failed (-84)
févr. 01 18:18:40 raspberrypi kernel: Bluetooth: hci0: Frame reassembly failed (-84)
févr. 01 18:18:40 raspberrypi kernel: Bluetooth: hci0: Frame reassembly failed (-84)
févr. 01 18:18:40 raspberrypi kernel: Bluetooth: hci0: Frame reassembly failed (-84)
févr. 01 18:18:40 raspberrypi bluealsa[340]: /usr/bin/bluealsa: Missing RTP packet: 90 != 89
févr. 01 18:18:41 raspberrypi bluealsa[340]: /usr/bin/bluealsa: SBC decoding error: No such file or directory
févr. 01 18:18:41 raspberrypi kernel: Bluetooth: hci0: Frame reassembly failed (-84)
févr. 01 18:18:41 raspberrypi kernel: Bluetooth: hci0: Frame reassembly failed (-84)
févr. 01 18:18:41 raspberrypi kernel: Bluetooth: hci0: Frame reassembly failed (-84)
```

Any idea of how to solve that ?

Many thanks in advance

nicokaiser, 2020-02-01:
See "Limitations" in the README.md:

> You might want to use a Bluetooth USB dongle or have the script disable Wi-Fi while connected (see `bluetooth-udev`), as the BCM43438 (Raspberry Pi 3, Zero W) has severe problems with both switched on, see [raspberrypi/linux/#1402](https://github.com/raspberrypi/linux/issues/1402).

Depending on your Bluetooth connection (and maybe on some other factors) this may improve the connection, however the internal Bluetooth remains crappy and using an external USB dongle really helps.

docgloucester, 2020-02-01:
As stated in the Edit line at the end of the thread, Wi-Fi is disabled by default on my RPi.
My home WI-Fi network operates fully on the 5GHz and I can't see anything obvious clogging up the 2.4GHz band that Bluetooth uses.

Is it really only due to the crappy nature of the RPi embedded Bluetooth equipement ? Wouldn't you see anything to do in the Bluetooth UART configuration for instance ?

nicokaiser, 2020-02-01:
This is discussed in raspberrypi/firmware#1150 (among many other places). It _may_ be related to high system load (whatever "high" means, maybe decoding A2DP packages and feeding built in sound card is enough), but after fiddling around many days, I decided to add a dedicated Bluetooth dongle and all those problems disappeared, and since I want to have high quality audio I also replaced the internal sound card by a I2S module (which is also worth it).

I did not try if the Raspberry Pi 4's chip does better, neither did I dive into UART configuration...

HiFiBerry OS does not even try to use BT on the Pi 3: https://github.com/hifiberry/hifiberry-os/blob/master/buildroot/package/btspeaker/checkpi.sh (Which also disables BT on the Pi 3 with dedicated USB dongles, but that's another story).

docgloucester, 2020-02-01:
Alright, that github comment sums up the problem pretty well : https://github.com/raspberrypi/linux/issues/2264#issuecomment-344911712

"The rev 1.2 Pi3B lacks the flow control signals to the Bluetooth modem (we ran out of pins), but the rev 1.3 board drops the BT PCM interface and hooks up the flow control. I think what you are seeing is occasional data loss when the FIFOs overflow, something which is hard to avoid in all circumstances. You may be able to lessen the problem by reducing the baudrate on the modem - try editing /usr/bin/btuart, replacing the 921600 with 460800 and rebooting."

It seems the 3B+ has the same problem described here for the first revisions of the 3B.

Reducing the baudrate as explained did the trick for me :)

closed 2020-02-01 by docgloucester

nicokaiser, 2020-02-01:
Good to know that there is this baudrate trick!

docgloucester, 2020-02-01:
Thank you very much for your efforts and your time 👍

amo13, 2021-04-11:
For anyone landing here, I can confirm that the `bluetooth-udev`-trick worked for me on the Pi4 B. No more stuttering or dropouts.
Just uncomment `#ifconfig wlan0 down &` and `#ifconfig wlan0 up &` in `/usr/local/bin/bluetooth-udev`.

nicokaiser, 2021-04-11:
I can confirm this trick works for _some_ environments, however in general (especially with Bluetooth distances >1m or when there are many Wi-Fi networks around) the Raspberry Pi internal Bluetooth module is _not_ capable enough.