nicokaiser/rpi-audio-receiver #40: spotifyd not working

kromsam, 2020-03-05:
I have run the script on my Raspberry Pi Zero W with a Phat DAC.
I have installed all possible options.

For some reason spotifyd is not working. I filled in the config files but it does not show up in my other Spotify clients.

In the installation script this is what happens:
```
Do you want to install Spotify Connect (spotifyd)? [y/N] y
spotifyd
```
And then nothing happens, the script just continues to the next install option. The configuration file is there and I can edit it. But when I run `spotifyd`, I get:
```
Illegal instruction
```
I saw this: https://github.com/Spotifyd/spotifyd/issues/259/ But I didn't know what to make of it.

kromsam, 2020-03-05:
I saw that I've to change a config file.

https://github.com/nicokaiser/rpi-audio-receiver/commit/fb90f5e25673a117ceeae6d568ae9eca20036881

Maybe it would be handy if one could choose this option during installation. A reference to this in the readme would be convenient for all Zero users!

Awbmilne, 2020-08-25:
Pull request worked for my Raspberry Pi Zero w. Thanks!

nicokaiser, 2020-09-20:
Fixed in 353a0d462388cceca3357a419d3e5bca05468944: the architecture is now auto-selected on installation.

closed 2020-09-20 by nicokaiser