nicokaiser/rpi-audio-receiver #44: Question regarding enable-script for other cards

voideater, 2020-05-11:
Hey,

first of all thanks for the git repo, it sure made making the my rpi 3b+ to a audio receiver much easier than I anticipated!

I don't got a ``Hifiberry`` card, but a ``Allo Digione``  card that basically does the same. It's working fine now with the rpi and I've gotten it working for the services I want (Bluetooth receiver and Spotify connect). 

I only run a few of the scripts from your git to have better control / try improve my understanding a bit. Obviously one of the scripts I didn't run was the ``enable-hifiberry.sh`` script as I don't have any of those cards, but I was thinking maybe I should do something similar but adapt it to the card I got. 

I see that our either create or alter the ``/etc/asound.conf`` file. On my device (latest Raspbian 10 ``buster``, full dist-upgrade last week) this file doesn't exist (at least not as ``/etc/asound.conf``, but got a similar file in ``/usr/share/alsa/alsa.conf`` - are these the same and is this the one that should be edited?). 

Does this part of the process have a noticeable effect on the final result? And if so, are the values in the script device-specific for the Hifiberry products or just for the rpi device (I see the comments have different values for rpi zero, i.e.).

The Allo Digione card is enabled using ``dtoverlay=allo-digione``, but when running other commands such as ``aplay -l`` or ``cat /proc/asound/cards`` it displays slightly modified names for the card, so I'm not sure which would be correct to use if I were to try making my own ``/etc/asound.conf`` file. See below for outputs.

```
pi@raspberrypi:~ $ cat /proc/asound/cards
 0 [sndallodigione ]: RPi-WM8804 - snd_allo_digione
                      snd_allo_digione
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sndallodigione [snd_allo_digione], device 0: Allo DigiOne HiFi wm8804-spdif-0 [Allo DigiOne HiFi wm8804-spdif-0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
```
Do you think something as simple as changing ``hifiberry`` with ``sndallodigione`` in the parts of your script that you output to ``/etc/asound.conf``  would be sufficient?

I know I probably shouldn't tweak more as it's working now, but I just have a feeling I might miss out on something so hard to resist! ;)

nicokaiser, 2020-05-15:
The `/etc/asound.conf` file may not exist in default installations, that's why you need to create it if you would like to change the default config (which is in `/usr/share/alsa/alsa.conf`).

In the `asound.conf` provided in `enable-hifiberry.sh` the `hifiberry` name is only a name (alias) for the first audio device (type hw, card 0, device 0). So as you already set up your DigiOne correctly, it should work and be available as first audio device. You can use the `asound.conf` without modifications, this should work. If you want, you can change the `hifiberry` to anything you want (e.g. `allodigione`), as long as it's the same in both places.

Hope I could help you!

closed 2020-05-15 by nicokaiser

voideater, 2020-05-15:
Ah, that makes sense then. Thanks a lot for the explanation, and again thanks for the scripts!