nicokaiser/rpi-audio-receiver #48: connection pin

OmegaMatze, 2020-05-17:
Hallo,

i'm glad for your work - installation was easy and is working.
I have only one question: how can I enable a PIN or somthing to restrict who can connect to the Pi via Bluetooth?

many Thx!

nicokaiser, 2020-05-24:
That's a nice idea! Maybe this could be done by checking the GPIO pins periodically by a script (or even only at startup?) and if a jumper is set, then make the device discoverable via `bluetoothctl`. 

Maybe something like the read/write PIN in this script: https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/master/read-only-fs.sh

ToeffToeff, 2020-05-26:
Hi, I'm not sure if @OmegaMatze meant a physical GPIO pin on the Pi. Sounds more like he's looking for a 4 digit pin for bluetooth authentication like a password.

Anyways, I like both ideas :)

nicokaiser, 2020-08-08:
Oh, I see. I think requiring a Bluetooth PIN is not really supported anymore with the "Secure Simple Pairing Mode" (set in `/etc/systemd/system/bthelper@.service.d/override.conf`). I did not get any connection working without sspmode...

JasonLG1979, 2020-08-29:
It might be possible with ```bt-agent``` it does have a ```-p, --pin``` https://manpages.debian.org/buster/bluez-tools/bt-agent.1.en.html

Although I'm not sure how it would work in ```NoInputNoOutput``` since we can't provide any feedback? the manpage says:

>if the key matches, otherwise it will request the user for manual authorization.

nicokaiser, 2020-09-02:
I just added a patch that switches from using an own agent implementation to `bt-agent`. Also, some instruction on how to enable PIN code pairing. For my setup (with USB Bluetooth dongle) this seems to work well, can anybody test this with the internal Raspberry Bluetooth module?

closed 2020-09-02 by nicokaiser

JasonLG1979, 2020-09-02:
The pin can be up to 6 digits right? If so it might be better for the example to be ```123456```?

JasonLG1979, 2020-09-02:
The docs for the ```DisplayPinCode``` method say this:

```Note that the PIN will always be a 6-digit number zero-padded to 6 digits. This is for harmony with the later specification.```

I know that is method is for verifying the pin from another device but my thought is what's good for the goose is good for the gander.

JasonLG1979, 2020-09-02:
You might also add that an easy way to get your Bluetooth mac address is to use ```hciconfig -a```.

nicokaiser, 2020-09-03:
Note that connections without SSP are _not_ supported on iOS devices. I could not find a way to enable pairing with SSP and PIN.

nicokaiser, 2020-09-03:
I reverted the docs about PIN pairing. It just does not work.

- iOS refuses to pair with devices that do not support SSP. So sspmode needs to be on
- Event with sspmode=1 and a PIN file, iOS would not connect at all
- macOS allows pairing with devices with sspmode=0 (and a PIN configured on the Pi). When sspmode=1, macOS decides upon a PIN and either the Pi has a fixed PIN list (which does not match => connection refused), or requires keyboard input on the Pi (which is not desired for a headless device).

So, I'll close this and conclude, once again, that configuring Bluetooth on Linux is a mess and only works with an USB dongle, without a PIN code and thus without additional security (anyone can connect). Sorry.

JasonLG1979, 2020-09-03:
Predefined pins are really kinda a hack anyway. They don't really offer any sort of extra security really since they're so short they wold be crackable in about as fast as you could send 10000 guesses and there's no encryption involved so they can be sniffed. If someone is concerned about having a device that accepts all pairing requests they should just wire up a button to a couple GPIO pins like was mentioned previously. They make cheap button shims and hats.

 https://www.adafruit.com/product/3582

commit 3678cf9f (2020-09-02): docs: add instructions for Bluetooth pairing with PIN (fixes #48)