nicokaiser/rpi-audio-receiver #52: Uninstall the script

rahulonmars, 2020-05-24:
How do you uninstall everything that was installed by the script and restore to the earlier state?

nicokaiser, 2020-05-24:
This is not really supported.

This script is an easy way to install audio receiver software on a freshly installed Raspbian installation and is not meant to run on e.g. a heavily customized Raspbian installation. So, the easiest way is to just delete the SD card and flash a clean Raspbian image again.

The other way is to have a look at the scripts and try to revert everything by hand. There will not be an uninstaller or something like this, sorry.

closed 2020-05-24 by nicokaiser

rahulonmars, 2020-05-25:
Thank you for replying !
I'll have a look at the script and uninstall the components :)

Btw, beautiful product !