nicokaiser/rpi-audio-receiver #59: Bluetooth is off at reboot

spoutn1k, 2020-08-12:


nicokaiser, 2020-08-14:
Could you send the output of `sudo journalctl -u bthelper@hci0.service`? Maybe this is a timing issue (again), but I currently do not have a Raspberry Pi 4 to test this...

Ae3NerdGod, 2020-08-17:
[hci0journalctl.txt](https://github.com/nicokaiser/rpi-audio-receiver/files/5082067/hci0journalctl.txt)

thats my output, rpi4b 8gb, buster lite with mate installed (blueman not installed)

how would i go about starting the bluetooth manually?

nicokaiser, 2020-08-17:
Can you add one line (the one with `Type=oneshot` to `/etc/systemd/system/bthelper@.service.d/override.conf`, so it looks like this?

```
[Service]
Type=oneshot
ExecStartPost=/usr/bin/bluetoothctl discoverable on
ExecStartPost=/bin/hciconfig %I piscan
ExecStartPost=/bin/hciconfig %I sspmode 1
```

It seems like Raspbian's Bluetooth init scripts are a bit broken in this case...

spoutn1k, 2020-08-17:
Sorry for the delay, I do not have access to the device anymore. I ended up removing the `/etc/systemd/system/bthelper@.service.d/override.conf` file, and adding `Discoverable = true` in the `/etc/bluetooth/main.conf` file. This worked for my purposes, and I am sure the scans and sspmode can be set the same way. Unfortunately those configuration files are horribly documented, so I am not sure what the option is.

spoutn1k, 2020-08-17:
[Looking at the sources](https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/src/main.conf), there is definitely a way to set this up entirely via the config files. Might provide a pull request but I have no way of testing my changes.

nicokaiser, 2020-08-17:
I could not find anything about sspmode and discoverability in the BlueZ sources, neither did I find a way to reliably enable them other than calling `hciconfig hci0 sspmode 1`... But if you find a reliable solution, I'm very interested!

commit 748fe152 (2020-08-24): fix: fix bthelper startup order (fixes #59)

bthelper in Raspbian Buster is started as Type=simple, which causes ExecStartPost to be called to early.
closed 2020-08-24 by nicokaiser