nicokaiser/rpi-audio-receiver #63: Bluetooth improvements (discussion)

JasonLG1979, 2020-08-24:
I recently set up my Pi Zero as a Bluetooth receiver with bluealsa by reading and following several write ups and tutorials and bug reports scattered across the web. Your repo was invaluable but I believe your implementation could be improved. Here is a list of improvements that I have implemented. Before I put in any pull requests I wanted to get an idea of if you're interested in any of these?

1. bluealsa and bluealsa-aplay run under a unprivileged system user 'buealsa' who is a member of the audio and bluetooth groups. Adapted from [here](https://github.com/Arkq/bluez-alsa/wiki/Systemd-integration)

My service files look like this:

```
[Unit]
Description=Bluealsa daemon
Requires=bluetooth.service
After=bluetooth.service

[Service]
Type=dbus
BusName=org.bluealsa
User=bluealsa
Group=bluealsa
NoNewPrivileges=true
ExecStart=/usr/bin/bluealsa -i hci0 -p a2dp-sink --a2dp-force-audio-cd
Restart=on-failure
ProtectSystem=strict
ProtectHome=true
PrivateTmp=true
PrivateDevices=true
RemoveIPC=true
RestrictAddressFamilies=AF_UNIX AF_BLUETOOTH

[Install]
WantedBy=bluetooth.target
```

```
[Unit]
Description=Bluealsa audio player
Requires=bluealsa.service
After=bluealsa.service

[Service]
Type=simple
ExecStartPre=/bin/sleep 1
ExecStart=/usr/bin/bluealsa-aplay --profile-a2dp --single-audio --pcm-buffer-time=1000000 00:00:00:00:00:00
Restart=on-failure
ProtectSystem=strict
ProtectHome=true
PrivateTmp=true
RemoveIPC=true
RestrictAddressFamilies=AF_UNIX
User=bluealsa
Group=audio
NoNewPrivileges=true

[Install]
WantedBy=bluetooth.target
 ```

2. I noticed what I think is a bug. But if 2 devices are connected and one disconnects bluetooth becomes discoverable even though there is still a connected device. I racked my brain on this this one and this is the best solution I came up with.  bluetoothctl info spits out 'Missing device address argument' if no devices are connected so I use that to make sure nothing is connected before I enable discovery. (I also made my own sound for discovery mode).

My ```/usr/local/bin/bluetooth-udev``` looks like this:

```
#!/bin/bash
if [[ ! $NAME =~ ^\"([0-9A-F]{2}[:-]){5}([0-9A-F]{2})\"$ ]]; then exit 0; fi
action=$(expr "$ACTION" : "\([a-zA-Z]\+\).*")
if [ "$action" = "add" ]; then
    bluetoothctl discoverable off
    if [ -f /usr/local/share/sounds/__custom/device-added.wav ]; then
        aplay -q /usr/local/share/sounds/__custom/device-added.wav
    fi
    ifconfig wlan0 down &
fi
if [ "$action" = "remove" ]; then
    if [ -f /usr/local/share/sounds/__custom/device-removed.wav ]; then
        aplay -q /usr/local/share/sounds/__custom/device-removed.wav
    fi
    deviceinfo=$(bluetoothctl info)
    if [ "$deviceinfo" = "Missing device address argument" ]; then
        if [ -f /usr/local/share/sounds/__custom/discoverable.wav ]; then
            aplay -q /usr/local/share/sounds/__custom/discoverable.wav
        fi
        bluetoothctl discoverable on
        ifconfig wlan0 up &
    fi
fi
```

3. Without the ```--single-audio``` bluealsa-aplay arg it's possible for multiple devices to stream audio at the same time. I'm not sure if this is intentional but in my mind only one device should be able to stream at a time.  ```--single-audio``` doesn't stop multiple devices from being connected you can just pause or stop one stream and start the other.

4. python-dbus has been deprecated for YEARS and should not be used. I plan on rewriting it in PyGObject (I know my way around dbus and PyGObject)

JasonLG1979, 2020-08-24:
A short audio trouble shooting section would also be nice. 99.9% of the time running ```aplay -l``` and changing ```defaults.ctl.card 0``` and ```defaults.pcm.card 0``` in ```/usr/share/alsa/alsa.conf``` to whatever card you want to be default solves the problem of no sound.

JasonLG1979, 2020-08-25:
OK here is a rewrite of a2dp-agent.py in python 3 with no python-dbus. The only dependencies are ```python3``` and ```python3-gi```. It is functionally identical to the previous version. But does not use a crusty old version of python and the deprecated python-dbus garbage.  

```python3
#!/usr/bin/env python3

# Copyright (c) 2020 Jason Gray <jasonlevigray3@gmail.com>
# This program is free software: you can redistribute it and/or modify
# it under the terms of the GNU General Public License as published by
# the Free Software Foundation, either version 3 of the License, or
# (at your option) any later version.
# This program is distributed in the hope that it will be useful,
# but WITHOUT ANY WARRANTY; without even the implied warranty of
# MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the
# GNU General Public License for more details.
# You should have received a copy of the GNU General Public License
# along with this program. If not, see <http://www.gnu.org/licenses/>.

from gi.repository import Gio, GLib

AGENT_IFACE = 'org.bluez.Agent1'
AGENT_PATH = '/test/agent'

class Server:
    def __init__(self, con, path):
        method_outargs = {}
        method_inargs = {}
        for interface in Gio.DBusNodeInfo.new_for_xml(self.__doc__).interfaces:
            for method in interface.methods:
                method_outargs[method.name] = '({})'.format(''.join([arg.signature for arg in method.out_args]))
                method_inargs[method.name] = tuple(arg.signature for arg in method.in_args)
            con.register_object(object_path=path, interface_info=interface, method_call_closure=self._on_method_call)
        self._method_inargs = method_inargs
        self._method_outargs = method_outargs

    def _on_method_call(self, connection, sender, object_path, interface_name, method_name, parameters, invocation):
        args = list(parameters.unpack())
        for i, sig in enumerate(self._method_inargs[method_name]):
            if sig == 'h':
                args[i] = invocation.get_message().get_unix_fd_list().get(args[i])
        try:
            if self._method_outargs[method_name] != '()':
                invocation.return_value(GLib.Variant(out_args, (getattr(self, method_name)(*args),)))
            else:
                invocation.return_value(None)
        except:
            pass

class Agent(Server):
    """
    <!DOCTYPE node PUBLIC "-//freedesktop//DTD D-BUS Object Introspection 1.0//EN"
    "http://www.freedesktop.org/standards/dbus/1.0/introspect.dtd">
    <node>
        <interface name="org.bluez.Agent1">
            <method name="Release" />
            <method name="RequestPinCode">
                <arg direction="in" type="o" />
                <arg direction="out" type="s" />
            </method>
            <method name="DisplayPinCode">
                <arg direction="in" type="o" />
                <arg direction="in" type="s" />
            </method>
            <method name="RequestPasskey">
                <arg direction="in" type="o" />
                <arg direction="out" type="u" />
            </method>
            <method name="DisplayPasskey">
                <arg direction="in" type="o" />
                <arg direction="in" type="u" />
                <arg direction="in" type="q" />
[... 78 lines cut ...]
        Gio.BusType.SYSTEM,
        Gio.DBusProxyFlags.DO_NOT_LOAD_PROPERTIES,
        None,
        'org.bluez',
        '/org/bluez',
        'org.bluez.AgentManager1',
        None
    )
    manager.call_sync(
        'RegisterAgent',
        GLib.Variant('(os)', (AGENT_PATH, 'NoInputNoOutput')),
        Gio.DBusCallFlags.NONE,
        -1,
        None
    )
    manager.call_sync(
        'RequestDefaultAgent',
        GLib.Variant('(o)', (AGENT_PATH,)),
        Gio.DBusCallFlags.NONE,
        -1,
        None
    )
    mainloop = GLib.MainLoop()
    mainloop.run()
```

JasonLG1979, 2020-08-25:
Well doing a little debugging revealed that none of the methods in the agent are even called in ```NoInputNoOutput``` mode so it's pointless to even use it. ```bluez-tools`` has ``` bt-agent``` built into it that's written in C that does the exact same thing (basically nothing) you can start it with ```bt-agent --capability=NoInputNoOutput``` So you can ditch the ```python-gobject``` and ```python-dbus``` dependencies all together.

nicokaiser, 2020-08-27:
Thanks @JasonLG1979 for the great suggestions!

The solutino with `bt-agent` looks great, would be awesome to finally get rid of `a2dp-agent.py` and its dependencies! I did not have time to test it, but it looks promising!

As for the `--single-audio` argument: currently it is possible to play with multiple devices simultaneously, because either the ALSA device supports hardware mixing, or the `dmix` plugin is used. I thought this behaviour avoids quite some complexity when it comes to switching between sources. Projects like HiFiBerryOS only allow one source at once (with the benefit to be able to use the ALSA device exclusively) but take quite some effort to achieve this, especially when also AirPlay and Spotify is used.

Unprivileged user for `bluealsa`: I do not really think this is necessary: the only purpose of the device is playing audio, so event if it becomes possible to gain root access (due to a security hole), the device is compromised anyway. So I'd like to avoid the complexity of adding and managing users apart from what Debian brings along with its packages...

JasonLG1979, 2020-08-27:
>The solutino with bt-agent looks great, would be awesome to finally get rid of a2dp-agent.py and its dependencies! I did not have time to test it, but it looks promising!

I've been playing with ```bt-agent``` for a couple days and it works as intended but it hangs on shutdown for several minutes on my pi zero. It predates systemd and doesn't seem to play well with it as far as gracefully shutting down when asked. I'm in the process of re-rewriting ```a2dp-agent.py``` and incorperating ```python-systemd``` to provide tight systemd integration.

> As for the --single-audio argument: currently it is possible to play with multiple devices simultaneously, because either the ALSA device supports hardware mixing, or the dmix plugin is used. I [...]

```--single-audio``` shouldn't effect any other service. It only makes sure that only one bluetooth stream is playable at a time.

>Unprivileged user for bluealsa: I do not really think this is necessary: the only purpose of the device is playing audio, so event if it becomes possible to gain root access (due to a security hole), the device is compromised anyway.

I'm going to have to disagree with you on this. Creating a bluetooth device that accepts all pairs and that is potentially connected to your network is a huge security hole in and of itself. It's not a matter of if the pi can be compromised or even if you care that it is. It's a matter of the fact that the pi can be used as a stepping stone into your otherwise secure network. IMHO the bluetooth service shouldn't even be ran as root in this situation. There is already bluetooth and audio groups by default it's trivial to create an unprivileged system user and add them to just those groups. If you insist on not following my advice then at least disable bluetooth networking with  ```--noplugin=network``` in the bluetooth.service unit file.

JasonLG1979, 2020-08-27:
[Here](https://gist.github.com/JasonLG1979/3bcbd8300d144a4fa38a8e3c578c6dd3) is a rewrite of a2dp-agent.py with systemd integration.

The dependencies are  ```python3 python3-gi python3-systemd``` on bluetooth 2.1 or higher devices it's basically a noop (nothing in the agent class is ever called) because of "Just works" pairing but for pre 2.1 devices it will reject any attempt to use any service but A2DP.

To use this you'll need to add ``` Type=notify``` in the ``` [Service]``` section of ```a2dp-agent.service``` unit file because though the magic of ```python3-systemd``` ```a2dp-agent.py``` tells systemd when it's ready and will gracefully exit and pass error messages on failure that can be read with ```systemctl status a2dp-agent.service```.

nicokaiser, 2020-08-27:
I'd like to throw `bt-agent` into the discussion again: it does just this (be a simple Bluetooth agent that does `NoInputNoOutput` and does not have any dependencies. This eliminates the need of Python on the system, is pre-compiled and already supplied with Debian/Raspberry Pi OS.

I added this in #64. The shutdown problems you mentioned is because of a bug in the signal handlers, see khvzak/bluez-tools#30. The systemd unit in the PR works around this.

JasonLG1979, 2020-08-27:
> I'd like to throw `bt-agent` into the discussion again: it does just this (be a simple Bluetooth agent that does `NoInputNoOutput` and does not have any dependencies. This eliminates the need of [...]

If that does the trick then I say use it.

JasonLG1979, 2020-08-27:
Next up is ```bthelper```, specifically ```/usr/bin/bthelper``` it's a shell script that doesn't really seem to do anything on my pi zero (because I use it's on board bt) except turn bluetooth on and off again and cause a ```Failed to set privacy: Rejected (0x0b)``` error in the ```bluetooth.service```.

The script states this in the comments about cycling on and off:
```shell
# Force reinitialisation to allow extra features such as Secure Simple Pairing
# to be enabled

```
After nooping the script with ```exit 0``` I can confirm that it's pretty much useless as Simple Pairing is enabled by default anyway and you set ```sspmode 1``` (I'm still not sure what that does? The hciconfig docs kinda suck...)

```shell
hciconfig hci0 sspmode
hci0:	Type: Primary  Bus: UART
	BD Address: B8:27:EB:5B:28:8E  ACL MTU: 1021:8  SCO MTU: 64:1
	Simple Pairing mode: Enabled
```
That is without your ```bthelper@.service.d/override.conf```

JasonLG1979, 2020-08-27:
```sspmode 1``` I guess just enables Secure Simple Pairing so that's redundant also as as I've said it's enabled by default.

JasonLG1979, 2020-08-27:
```hciconfig %I piscan``` is also redundant as both ```PSCAN ISCAN``` are also enabled by default.

JasonLG1979, 2020-08-27:
```shell
hciconfig hci0 -a
hci0:	Type: Primary  Bus: UART
	BD Address: B8:27:EB:5B:28:8E  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING PSCAN ISCAN 
	RX bytes:816 acl:0 sco:0 events:56 errors:0
	TX bytes:3027 acl:0 sco:0 commands:56 errors:0
	Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH SNIFF 
	Link mode: SLAVE ACCEPT 
	Name: 'PiZero'
	Class: 0x040428
	Service Classes: Rendering
	Device Class: Audio/Video, HiFi Audio Device
	HCI Version: 4.1 (0x7)  Revision: 0x1ed
	LMP Version: 4.1 (0x7)  Subversion: 0x2209
	Manufacturer: Broadcom Corporation (15)
```

I've set my device class to 0x040428 if that matters.

nicokaiser, 2020-08-27:
Did you check if `sspmode` is also enabled by default on external (USB) Bluetooth dongles? The internal Bluetooth chip is really really not suitable for audio transmission (on any Raspberry Pi, I wasted so much time trying, it sucks), so it needs to work with dongles.

JasonLG1979, 2020-08-27:
> Did you check if `sspmode` is also enabled by default on external (USB) Bluetooth dongles?

No I have not but from what I understand Bluez 5 enables it by default if the device supports it.

>The internal Bluetooth chip is really really not suitable for audio transmission (on any Raspberry Pi, I wasted so much time trying, it sucks), so it needs to work with dongles.

I've had no problems on a pi zero after I increased the  pcm-buffer-time to 1000000.

JasonLG1979, 2020-08-27:
On a zero you can also try disabling the serial console, overclocking it, setting the cpu governor to performance, and renicing bluetooth related stuff. In my experience it's more about the transmitting device. My computer and phone do really well with the zero stock. My 5th gen fire 7 (with LineageOS 12.1 (android 5.1)) drops out from time to time unless I tweak the zero a bit.

JasonLG1979, 2020-08-28:
Here are a couple other suggestions:

1. Disable unneeded bluetooth plugins in the bluetooth.service file with --noplugin=sap,network,hog,health,midi. The sap plugin in particular causes bluetoothd to throw an error.

2. set the Type to dbus and the BusName to org.bluealsa in the bluealsa service file to stop bluealsa-aplay from throwing an error about not finding the busname in any service file and not being able to find devices.

nicokaiser, 2020-08-28:
I'll add the Type=dbus enhancements, this seems like a good thing that may help  on timing issues on startup. Maybe I'll add the User/Group settings commented out (like the in the BlueALSA docs), so people can opt-in by themselves (as this involves quite some steps to achieve, including changing the dbus configuration, I'd like to keep this as optional).

JasonLG1979, 2020-08-29:
```--noplugin=sap``` is also an easy way to work around a startup bug in bluetoothd unless you plan on accessing your phone's SIM card with your Raspberry Pi,lol!!!

https://en.wikipedia.org/wiki/List_of_Bluetooth_profiles#SIM_Access_Profile_(SAP,_SIM,_rSAP)

pucgenie, 2020-08-31:
> OK here is a rewrite of a2dp-agent.py in python 3 with no python-dbus. The only dependencies are `python3` and `python3-gi`. It is functionally identical to the previous version. But does not use a crusty old version of python and the deprecated python-dbus garbage.
> 
> ```python
> #!/usr/bin/env python3
> 
> # Copyright (c) 2020 Jason Gray <jasonlevigray3@gmail.com>
> # This program is free software: you can redistribute it and/or modify
> # it under the terms of the GNU General Public License as published by
> # the Free Software Foundation, either version 3 of the License, or
> # (at your option) any later version.
> # This program is distributed in the hope that it will be useful,
> # but WITHOUT ANY WARRANTY; without even the implied warranty of
> # MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the
> # GNU General Public License for more details.
> # You should have received a copy of the GNU General Public License
> # along with this program. If not, see <http://www.gnu.org/licenses/>.
> 
> from gi.repository import Gio, GLib
> 
> AGENT_IFACE = 'org.bluez.Agent1'
> AGENT_PATH = '/test/agent'
> 
> 
> class Server:
>     def __init__(self, con, path):
>         method_outargs = {}
>         method_inargs = {}
>         for interface in Gio.DBusNodeInfo.new_for_xml(self.__doc__).interfaces:
>             for method in interface.methods:
>                 method_outargs[method.name] = '({})'.format(''.join([arg.signature for arg in method.out_args]))
>                 method_inargs[method.name] = tuple(arg.signature for arg in method.in_args)
>             con.register_object(object_path=path, interface_info=interface, method_call_closure=self._on_method_call)
>         self._method_inargs = method_inargs
>         self._method_outargs = method_outargs
> 
>     def _on_method_call(self, connection, sender, object_path, interface_name, method_name, parameters, invocation):
>         args = list(parameters.unpack())
>         for i, sig in enumerate(self._method_inargs[method_name]):
>             if sig == 'h':
>                 args[i] = invocation.get_message().get_unix_fd_list().get(args[i])
>         try:
>             if self._method_outargs[method_name] != '()':
>                 invocation.return_value(GLib.Variant(out_args, (getattr(self, method_name)(*args),)))
>             else:
>                 invocation.return_value(None)
>         except:
>             pass
> 
> 
> class Agent(Server):
>     """
>     <!DOCTYPE node PUBLIC "-//freedesktop//DTD D-BUS Object Introspection 1.0//EN"
>     "http://www.freedesktop.org/standards/dbus/1.0/introspect.dtd">
>     <node>
>         <interface name="org.bluez.Agent1">
>             <method name="Release" />
>             <method name="RequestPinCode">
>                 <arg direction="in" type="o" />
>                 <arg direction="out" type="s" />
>             </method>
>             <method name="DisplayPinCode">
>                 <arg direction="in" type="o" />
>                 <arg direction="in" type="s" />
>             </method>
>             <method name="RequestPasskey">
>                 <arg direction="in" type="o" />
>                 <arg direction="out" type="u" />
>             </method>
>             <method name="DisplayPasskey">
>                 <arg direction="in" type="o" />
[... 85 lines cut ...]
>         '/org/bluez',
>         'org.bluez.AgentManager1',
>         None
>     )
>     manager.call_sync(
>         'RegisterAgent',
>         GLib.Variant('(os)', (AGENT_PATH, 'NoInputNoOutput')),
>         Gio.DBusCallFlags.NONE,
>         -1,
>         None
>     )
>     manager.call_sync(
>         'RequestDefaultAgent',
>         GLib.Variant('(o)', (AGENT_PATH,)),
>         Gio.DBusCallFlags.NONE,
>         -1,
>         None
>     )
>     mainloop = GLib.MainLoop()
>     mainloop.run()
> ```

Somehow (ssb) doesn't work - use (ssv).
> self._trust_variant = GLib.Variant('(ssv)', ('org.bluez.Device1', 'Trusted', GLib.Variant.new_boolean(True)))
Thanks @JasonLG1979 and @nicokaiser

JasonLG1979, 2020-08-31:
@pucgenie, That's weird. type v is a string type I though? Anyway it's neither here nor there. ```bt-agent``` does the same thing with less dependencies so that's my choice. Here is a step by step guide to how I set everything up.

https://github.com/JasonLG1979/PiZero-Bluetooth-Audio-Receiver

berkakinci, 2021-08-22:
> https://github.com/JasonLG1979/PiZero-Bluetooth-Audio-Receiver

Thanks @JasonLG1979.
I struggled with these repo scripts for a while, with no success in getting bluetooth enabled.  I see a lot of explanations and workarounds kicking around.  I didn't have the appetite to keep debugging after an hour or so.
Your repo, though only instructions, worked out of the box.  I think too much has changed since these scripts were written. I'm on Pi Zero W Raspberry Pi OS -- "2021-05-07-raspios-buster-armhf-lite"
Thanks again.  Now I'm finally off to customizing...

nicokaiser, 2021-12-18:
Since this little script now achieves Bluetooth connectivity using PulseAudio (instead of BlueALSA, which is not included in Raspberry Pi OS 11 anymore), I'm now closing the related tickets. Hope PulseAudio works better...

closed 2021-12-18 by nicokaiser