nicokaiser/rpi-audio-receiver #66: Upsampling provides no benefit

JasonLG1979, 2020-08-29:
In ```enable-hifiberry.sh``` you upsample the audio to ```192000```. Not only is this a waste of CPU cycles it actually degrades fidelity for audio that's sample rate is not an integer divisor of ```192000``` namely ```44100``` audio that accounts for the vast majority of audio that is played because by default ALSA resamples with a trivial algorithm (basically straight multiplication division with a lot of rounding errors because of the remainders). Changing the bit depth to ```S32_LE``` is a good idea because it gives more headroom for digital volume control though but resampling is no good. If you don't resample ALSA will just play whatever at it's native sampling rate. There's really no need to specify a global sampling rate. You can just use ```softvol``` in ALSA by itself for software volume without dmix.

nicokaiser, 2020-09-23:
I've read somewhere that the upsampling algorithm of ALSA is quite okay, and may even be (much) better than anything that is done in the hardware: I'm not sure what the "native" sampling rate of e.g. HiFiBerry's DAC+ series is, but I think they can play at any multiple of 44100 or 48000...

JasonLG1979, 2020-09-23:
Modern delta sigma DACs are going to oversample even 192000. That's just the max input sampling rate due to the Linux drivers for some HiFiBerry DACs. Delta sigma DACs work by upsampling the crap out of the PCM audio by as much as 64 times or more and then they convert it to Pulse Width Modulation and then they use filters to get rid of the ultrasonic noise. To defeat the upsampler in the PCM5122 chip used in the DAC+ Pro for example you'd have to oversample to 384000. It's impossible to defeat the resampling in the PCM1796 chip used in the DAC 2 HD. Even at 192000 it upsamples 32 times. The data sheets for both DAC chips are easy to find.

As far as the quality of upsampling algorithm(s) of ALSA. There are several you can choose from. The default if ```libasound2-plugins``` is not installed is linear, which creates all kinds of rounding errors when not resampling to values that are not divisible by each other. So going from 44.1 to any thing not 88.2, 176.4 and so on is going to degrade the sound quality. Even with a "good" algorithm upsampling in and of itself NEVER improves sound quality. You can't make something from nothing.

The only possible benefit of upsampling would to make up for the fact that the clock on the Pi's is not capable of producing a clean 44.1 signal. And that only affects DACs that don't provide their own master clock and even then the benefit is suspect at best.

The short answer is the default upsampling algorithm of ALSA sucks. You can't get to the "native" sampling rates of any HiFiBerry Dacs. Upsampling NEVER makes anything sound better by itself.

JasonLG1979, 2020-09-23:
As an experiment I invite you to install ```libasound2-plugins``` and add this to ```/etc/asound.conf``` to change the upsampling algorithm to something decent and then play some 44.1 audio upsampled to 192000 and see what happens:

 ```defaults.pcm.rate_converter "speexrate_best"```

Spoiler:

It will murder your CPU and you'll more than likely experience dropouts and have unusable audio.

JasonLG1979, 2020-09-24:
If it were me ```/etc/asound.conf``` would look like this, (in fact mine look very similar):

```
# defaults.pcm.card 0 # Redundant, 0 is already the default.
# defaults.ctl.card 0 # Redundant, 0 is already the default.

pcm.hifiberry {
    type hw
    card 0
    device 0
}

pcm.dmixer {
    type dmix
    ipc_key 1024
    ipc_perm 0666
    slowptr true # read the docs
    hw_ptr_alignment roundup # read the docs
    slave.pcm hifiberry # Don't need quotes for strings that don't contain special chars.
    slave {
        period_time 250000 # 0.25 secs.  
        buffer_time 1000000 # 1.0 secs. Low latency is really only a concern when recording. Big buffer equals less chance of dropouts and/or clicks and pops.
        rate 44100 
        format S16_LE
    }
    bindings {
        0 0
        1 1
    }
}

ctl.dmixer {
    type hw
    card 0
}

pcm.softvol {
    type softvol
    slave.pcm dmixer
    control {
      name Softvol
      card 0
    }
    min_dB -90.2 # not sure why this isn't -96.3? If you're going to go to -90.2 why not go for 16 bit full scale?
    max_dB 0.0
}

pcm.!default {
    type plug
    slave.pcm softvol
}
```

JasonLG1979, 2020-10-02:
Here if you want to be able to configure pretty much everything [this Gist](https://gist.github.com/JasonLG1979/cb8616862f62134790ffadbd73879cde) will let you do it by changing just a couple lines.

The part that a user would touch looks like this:
```
defaults.pcm.card 1 # Change to the card you want to be default.
defaults.ctl.card 1 # Change to the card you want to be default.

ctl.!default {
    type plug
    # slave.pcm "HQstereoHardWareVol20:CARD=0,DEV=0,SUBDEV=-1,RATE=44100,FORMAT=S16_LE,RATE_CONVERTER=samplerate_linear,PERIODS=4,PERIOD_TIME=125000"
    # slave.pcm "HQstereoSoftWareVol20:CARD=0,DEV=0,SUBDEV=-1,RATE=44100,FORMAT=S16_LE,RATE_CONVERTER=samplerate_linear,PERIODS=4,PERIOD_TIME=125000"
    slave.pcm "HQstereoHardWareVol20:CARD=1"
}

ctl.!default {
    type hw
    card 1 # Change to the card you want to be default.
    device 0 # Change to the device you want to be default.
    subdevice -1 # Change to the subdevice you want to be default.
}
```

CARD, DEV, SUBDEV, RATE, FORMAT, RATE_CONVERTER, PERIODS, and PERIOD_TIME are all configurable in one line and you can choose between hardware and software volume control. If any arg is omitted the default value will be used. The Gist is set up for my personal setup with a USB DAC with hardware volume control.

nicokaiser, 2020-10-10:
Maybe you are right, I just copied the settings that HiFiBerry use for their HiFiBerry OS: https://github.com/hifiberry/hifiberry-os/tree/master/buildroot/package/hifiberry-tools/conf 

I guess people who use an external sound card (I2S or USB) will most probably touch their asound.conf according to their preferences anyway, so I'd like to leave sane simple defaults here.

As I do not have the resources to maintain a knowledge base for ALSA configuration in this repository, I'll close this for now, maybe I can link to your asound.conf in the docs?

closed 2020-10-10 by nicokaiser

JasonLG1979, 2020-10-10:
The main reason "audiophiles" love to upsample is the same reason they love crazy hi res audio. Because of ignorance. More must equal better (in their minds at least). Upsampled DSD remasters of digital recordings that were originally stored as PCM or DSD64 is a great example. You can't make something from nothing, you can't retrieve detail that's not there. Digital audio is science and math. Objective quality is absolutely quantifiable. Does the output match the input and is the difference (distortion) audible?

Here is a link to the alsa plugin docs. They read like stereo instructions (in a bad way) and they're out of date. You can't really use them to write an asound.conf because if you follow them to the letter it won't work.

https://www.alsa-project.org/alsa-doc/alsa-lib/pcm_plugins.html

The asound.conf docs are even worse.

 https://alsa.opensrc.org/Asoundrc

It took me a few days of trial and error and Googling to figure out what little I know.