nicokaiser/rpi-audio-receiver #67: RPi3 Unable To Pair

silverbullettruck2001, 2020-09-02:
@nicokaiser I have been trying to use this on my Rpi3 without success. I have disabled wifi on the Rpi3 to account for the issues with both wifi and bt being enabled, but even with that I cannot get my device to pair. The device I am using is a TaoTronics TT-BA09 Transmitter. Here is a link to the manual. TT-BA09 Manual. I am using the transmitter in TX mode as I want to connect it to my non-bluetooth TV and stream audio from the TV to the RPi3. I put the device in pairing mode and it never pairs. If I try to connect a pair of bluetooth headphones to the TT-BA09, it works as would be expected and the audio plays. Do you have any suggestions on how I can get this device to pair to the RPi3? Let me know if there's anything I can provide to assist in troubleshooting this with me. Also, I should mention this is on a fresh re-imaged card. Thanks!

nicokaiser, 2020-09-02:
Did you try to connect to the RPi3 with another device, like a phone or something? Just to check out if it works at all...

If it works, we need to figure out what the TT-BA09 (as a sender) does differently than the other devices.

If it also does not work, we need to figure out what's wrong with the RPi3's configuration. It _should_ work, even with the internal Bluetooth, but it is highly recommended to use a dedicated Bluetooth dongle for this, as the Bluetooth chip on the Raspberry Pis are not very capable and almost certainly lead to problems.

silverbullettruck2001, 2020-09-02:
@nicokaiser , thanks for the response.  Much appreciated!  I did verify that I can connect my phone to the RPi3 just fine and it plays audio as expected.  

What can I provide that would indicate what the TT-BA09 does differently?  Let me know.

Thanks!

nicokaiser, 2020-09-02:
Can you check `journalctl -f` during pairing, if anything suspicious is logged there?

I also changed the script so it uses a different Bluetooth agent (the piece of service that handles new devices), maybe this one works better. If this also fails, you could try changing to PIN code pairing (also in the README.md), in case the TT-BA09 does not support Simple Secure Pairing mode...

silverbullettruck2001, 2020-09-02:
I checked `journalctl -f` and nothing was logged that was of interest.  If I paired it with my phone there was a bunch of items logged, so maybe the lack of logging is suspicious, but I don't know why it wouldn't even attempt to pair?

I re-flashed the card and installed the latest script as you recommended.  Now I get some additional logs.  Here's what I get:

`Sep 03 00:00:58 raspi-receiver sudo[657]:       pi : TTY=pts/0 ; PWD=/home/pi ; USER=root ; COMMAND=/bin/journalctl -f
Sep 03 00:00:58 raspi-receiver sudo[657]: pam_unix(sudo:session): session opened for user root by pi(uid=0)
Sep 03 00:01:24 raspi-receiver bt-agent[356]: Device: 0B-26-75-03-AC-5F (0B:26:75:03:AC:5F) for UUID 0000110d-0000-1000-8000-00805f9b34fb
Sep 03 00:01:27 raspi-receiver bluetoothd[474]: connect error: Connection refused (111)
Sep 03 00:01:27 raspi-receiver bluetoothd[474]: /org/bluez/hci0/dev_0B_26_75_03_AC_5F/fd3: fd(22) ready
`

While it is good to get something logged, I am not sure why the connection was refused.  I would attempt your suggestion of using a PIN, but I don't even know what PIN to use since I didn't need a PIN when I paired this device with my headphones.

Any ideas on what to try next?  Again, huge thanks to helping me figure this out!

nicokaiser, 2020-09-06:
Sorry, I don't have an idea what is going wrong here... especially if it works with other devices...

closed 2020-09-06 by nicokaiser

silverbullettruck2001, 2020-09-06:
@nicokaiser is there some logging or something that would give some sort of info on what is going on?

nicokaiser, 2020-09-06:
I honestly don't know. All my Bluetooth experience on Linux and especially on Raspbian is more or less "trial and error" as this is an extremely complex and the documentation (of the Bluez stack, but also of sending devices) is somewhat incomplete.

One thing that seems strange is the "Connection refused" error you are getting. Are you sure you enabled SSP mode, the device is SSP compatible (should be if it's newer than let's say 5 years...) and the bt-agent is running?

silverbullettruck2001, 2020-09-07:
I decided to try using an RPi 4 instead of the RPi 3 to see if it acted any differently.  I found out two things.  One, I was not correctly setting the TT-BA09 to pair in Tx mode.  Two, the RPi 4 pairs just fine compared to the RPi 3.  After I got it working on the RPi 4 I went back and tried to get it working on the RPi3, and was able to make it a little further.  I now get some "traffic" when it is trying to pair, but it still won't pair.  What I have now is that when I launch `bluetoothctl` it cycles through three statuses but never connects.  Here is what I see:
```
[NEW] Device 0B:26:75:03:AC:5F 0B-26-75-03-AC-5F
[CHG] Device 0B:26:75:03:AC:5F Connected: no
[DEL] Device 0B:26:75:03:AC:5F 0B-26-75-03-AC-5F
```
I tried stopping the bt-agent and bt-helper services to see if they were auto-restarting or something, but that didn't seem to be the case.  

Any thoughts on what I can do to get this corrected on the RPi 3?