nicokaiser/rpi-audio-receiver #68: Equalizer not working

palsbo, 2020-09-06:
Finaly something that works out of the box - thank you!
I am using a Pi3 with HiFiBerry Amp.
I tried to install an Equalizer like this [ADDING EQUALIZATION USING ALSAEQ](https://www.hifiberry.com/docs/software/guide-adding-equalization-using-alsaeq/) and it works when I test with this `speaker-test -c2 -twav -l7`
However; the equalizer is not afecting sound comming from the bluetooth. 
How can I change this?
Thanks in advance.
Erik

nicokaiser, 2020-09-23:
It seems like bluealsa-aplay uses a different ALSA device than the equalizer.
You could try to change the `ExecStart` line in `/etc/systemd/system/bluealsa-aplay.service` so it includes the ALSA device?

```
ExecStart=/usr/bin/bluealsa-aplay --pcm=equal --pcm-buffer-time=250000 00:00:00:00:00:00
```

nicokaiser, 2020-10-31:
Closed due to inactivity.

closed 2020-10-31 by nicokaiser

MattCost, 2020-11-19:
I'm trying the same thing.
Followed the setup from hifiberry, and tried many other configs, none seem to work with bluealsa.

My asound.conf matches the example from hifi berry website.
When using speaker-test to generate test tones, the alsa equalizer does have an effect on the sound.

I've modified the ExecStart line to use a specific device using the --pcm switch, but no matter what, the alsa equalizer is being skipped over.

I don't want to open an issue here since this really isn't a bug with the setup scripts, but wondering if you have any ideas on where to start troubleshooting this? What logs should i be looking at, or perhaps a default pcm is getting used somewhere.

1ecstasy, 2022-02-12:
Hello there,
I have the same problem. I tried to incorporate the hifiberry tutorial asound.conf into the my nearly freshly installed one. then I turned off all the frequency bands for root and non-root user. my setup still keeps playing though, so I don't think the equalizer is working.

My asound.conf looks like this:
```
defaults.pcm.card 0
defaults.ctl.card 0

pcm.hifiberry {
  type hw
  card 0
  device 0
}
pcm.dmixer {
  type dmix
  ipc_key 1024
  ipc_perm 0666
  slave.pcm "hifiberry"
  slave {
    period_time 0
    period_size 1024
    buffer_size 8192
    rate 44100
    format S32_LE
  }
  bindings {
    0 0
    1 1
  }
}
ctl.dmixer {
  type hw
  card 0
}
pcm.softvol {
  type softvol
  slave.pcm "dmixer"
  control {
    name "Softvol"
    card 0
  }
  min_dB -90.2
  max_dB 0.0
}
pcm.plugequal{
  type equal;
  slave.pcm "softvol";
}
pcm.equal{
  type plug;
  slave.pcm plugequal;
}
pcm.!default {
  type plug
  slave.pcm plugequal
}
ctl.equal {
  type equal;
}
```

I know this is not within the target of the scripts but similarly to MattCost I'd appreciate any type of hint.