nicokaiser/rpi-audio-receiver #70: Add FastConnectable = true option?

mzealey, 2020-09-20:
It always takes at least 5 seconds for my linux laptop to connect to the pi with the standard bluetooth config.

If I add `FastConnectable = true` to the pi it always connects <1s. Perhaps worth adding to your script?

nicokaiser, 2020-09-21:
I could not really find what FastConnectable exactly does, other than enabling faster connections and leading to higher power consumption (on which device? On the client also?)

In some cases it seems to lead to poor sound quality: https://erikdubois.be/installing-bose-quietcomfort-35-linux-mint-18/ (In this report it is used on the A2DP source however, in our case it's the A2DP sink).

mzealey, 2020-09-21:
I too don't understand what it does but probably sends out advertisements/beacons more quickly than normal or something?

nicokaiser, 2020-10-31:
I added a section about this in the Wiki: https://github.com/nicokaiser/rpi-audio-receiver/wiki/Bluetooth-troubleshooting

closed 2020-10-31 by nicokaiser