nicokaiser/rpi-audio-receiver #71: connection dropping

cvdwl, 2020-09-22:
First: many thanks for this repo!  I went through many iterations of blindly fumbling at StackOverflow suggestions before landing here.  I'm basically Linux capable, but the dark inner workings of Pulse, Alsa, and various flavors of Bluetooth are way out of my wheelhouse.

Disclaimer aside; when I try to connect or reconnect my Moto X4 as a BT sound source, sometimes it works, and sometimes it goes into a drop and reconnect cycle.  The only obvious logs I could find are (note, host and BT id obfuscated):

`Sep 22 08:15:11 foo kernel: [63356.142943] input: 00:00:00:00:00:00 as /devices/virtual/input/input4
Sep 22 08:15:21 foo kernel: [63366.393836] input: 00:00:00:00:00:00 as /devices/virtual/input/input5
Sep 22 08:15:37 foo kernel: [63382.155195] input: 00:00:00:00:00:00 as /devices/virtual/input/input6
Sep 22 08:15:52 foo kernel: [63397.151423] input: 00:00:00:00:00:00 as /devices/virtual/input/input7
Sep 22 08:18:45 foo kernel: [63570.139936] input: 00:00:00:00:00:00 as /devices/virtual/input/input8`

Rebooting the phone (between 7 and 8) allowed a reconnect.

I realize this is likely a driver/hardware issue (possibly on the phone side) but would welcome any guidance.

Thanks again for this.  In general it works really well!

nicokaiser, 2020-09-23:
Hello @cvdwl! Two questions, 

1. Does it work at all? So, can you play music from your phone to the Raspi (at least for some time until the connection drops)?
2. Which Bluetooth hardware do you use? A Bluetooth USB dongle or the internal one of the Raspi?

In case it does work sometimes it seems more like a hardware issue, and if you use the Raspi 3 Bluetooth chip, you might consider getting a cheap USB Bluetooth dongle. The internal Bluetooth of the Raspberry Pi (all models) is very limited and unstable, especially when using audio. For some people it works slightly better when disabling Wi-Fi (at least while a BT device is connected, see `/usr/local/bin/bluetooth-udev` how to achieve this), but all in all it's not fun at all.

JasonLG1979, 2020-09-23:
Disable the triggerhappy service and see if that helps.

nicokaiser, 2020-10-31:
Closed due to inactivity.

closed 2020-10-31 by nicokaiser

cvdwl, 2020-12-01:
Sorry, was offline for a while.  The problem appears to be much deeper than mentioned.  One app (Radio Paradise) on one phone (Moto X4) doesn't work.  Basic workarounds exist.

Awbmilne, 2020-12-01:
As @nicokaiser Referenced with on the Bluetooth/Wifi

I had a similair issue with audio dropping in and out on my Pi Zero. This was due to interferance between the Wifi and Bluetooth on the Pi Zero chip. It was solved by allowing the Bluetooth Module to Disable Wifi when a device is connected over bluetooth. This has worked for me so far and hasnt gotten stuck with Wifi disabled or had audio dropout.

This issue is referenced in the [README: Limitations Section](https://github.com/nicokaiser/rpi-audio-receiver/blob/f594f4e1510479fbe50808fd3e8d21c15789c5e4/README.md#limitations)\
NicoKaiser Casually added a solution to this in d6d6c17e7cd267e32d0f0bc6743350a173d4dd45

The solution (at least for me) is to uncomment 2 lines in the Bluetooth install script:
- [Line 92](https://github.com/nicokaiser/rpi-audio-receiver/blob/f594f4e1510479fbe50808fd3e8d21c15789c5e4/install-bluetooth.sh#L92)
- [Line 97](https://github.com/nicokaiser/rpi-audio-receiver/blob/f594f4e1510479fbe50808fd3e8d21c15789c5e4/install-bluetooth.sh#L97)

These were intentionally left for this purpose. Then reinstall.

If you dont want to reinstall, just edit the config file `/usr/local/bin/bluetooth-udev`\
Uncomment lines _9_ and _14_