nicokaiser/rpi-audio-receiver #74: Add a wiki

JasonLG1979, 2020-09-23:
It would be helpful to add a wiki with the locations to the different services config files along with example configs and the default values of those configs.

Basic audio troubleshooting would also be helpful.

nicokaiser, 2020-10-10:
I like the idea!

JasonLG1979, 2020-10-10:
> I like the idea!

You have to enable the wiki before anyone can add anything to it.

closed 2020-10-31 by nicokaiser

JasonLG1979, 2020-11-11:
You also have to allow other people to edit it otherwise I can't add anything to it.

nicokaiser, 2020-11-13:
You are right. I changed the permissions to public, let's see if this works (I'd like to keep an eye on the structure and the contents of the Wiki and not let it become an all-purpose-"Audio on Linux" wiki, but maybe it works).

JasonLG1979, 2020-11-13:
> I'd like to keep an eye on the structure and the contents of the Wiki and not let it become an all-purpose-"Audio on Linux" wiki, but maybe it works

Well this repo has the potential to enable several different audio services. Because of that my general idea was to take them though the steps of creating a functional default output. A default that will take pretty much anything you'd throw at it and make noise. ALSA is ALSA regardless of distro so 99.9% of it would be by venture of that fact "all purpose".

JasonLG1979, 2020-11-13:
There I added an audio configuration page. I think it's pretty simple but again by general necessity it's pretty "all purpose" in that generally speaking it's pretty distro agnostic. If you don't like it ofc feel free to delete and/or edit it.

commit fa3cd283 (2020-10-31): docs: add link to the wiki (fixes #74)