nicokaiser/rpi-audio-receiver #75: further integration to control sender device

Thomas-GeDaD, 2020-10-05:
Hi,
is there a way to control the sender device via the reciever? 
I want to intigrate the reciever in an marine aplication and want to control the sending device through a nmea2000 conection.
So is it possible to connect so that it is possible to get control for next track, track back, pause, stop, start and so on?? Further get the track list?

nicokaiser, 2020-10-31:
I'm not sure if this is possible with BlueZ... at least it is not in the default configuration.

To control the device, AVRCP is needed (currently this project only supports A2DP as it is built to be a "dumb" audio receiver with no buttons etc.).

You could try to play with the BlueZ 5 D-Bus interface, as mentioned here: https://askubuntu.com/questions/591995/avrcp-with-bluez-how-to-control-audio-on-a-phone-from-ubuntu

nicokaiser, 2021-12-18:
Closing this since we need to move to PulseAudio (Debian 11 / Raspberry Pi OS 11).

closed 2021-12-18 by nicokaiser