nicokaiser/rpi-audio-receiver #76: Windows 10 will not play (Bluetooth nor UPnP)

DaCapitalist, 2020-10-24:
Hi!

First of all many thanks for this handy script! That saved me a lot of time.
I installed Shairport, BlueALSA and UPnP on my Raspberry Pi Zero WH.

Everything works fine from my iPhone, but my Windows 10 laptop (Asus N53SV) refuses to connect and play music via the Pi.
It can detect the UPnP server but cannot play via "play to" giving the message "unexpected device error" (translated from German).
When I go to "Bluetooth and other Devices" in the setting and scan for any of the 3 types of devices it will not see/find my BlueALSA server at all.

Any idea of what I am doing wrong?

Many thanks!

nicokaiser, 2020-11-22:
Can you have a look if you find anything in the device logs (`sudo journalctl`, e.g.)? Maybe there are some hints on what is going wrong... Does connecting from other devices (mobile phone via Bluetooth) work?

DaCapitalist, 2020-12-09:
Not sure how I did it but it seems to at least connect via Bluetooth now when using Windows. Unfortunately,  I need to make Windows forget the device each time before reconnecting, because it will not connect to an already paired Raspberry Pi. This I find strange but did not have the chance to review the logs yet. Will certainly do. Any idea what could be the reason?

closed 2021-12-18 by nicokaiser