nicokaiser/rpi-audio-receiver #81: Could not acquire an MPRIS interface on read only Raspberry OS

giovi321, 2020-11-12:
Hi all,
I keep getting the following error when trying to start shairport-sync 3.3.5-1~bpo10+1 (installed through apt) on Raspberry OS Linux 5.4.72-v7+:
`Could not acquire an MPRIS interface named "org.mpris.MediaPlayer2.ShairportSync.i1175" on the system bus.`
I can usually fix the problem by issuing:
```
systemctl daemon-reload
systemctl restart shairport-sync
```
I had a similar issue with the following error, maybe it could hbe of similar nature?
`Could not acquire a Shairport Sync native D-Bus interface "org.gnome.ShairportSync.i1007" on the system bus.`
that I have solved by copying the file `scripts/shairport-sync-dbus-policy.conf` to `/etc/dbus-1/system.d`.

nicokaiser, 2020-11-22:
That's strange, this installation script should install a custom shairport-sync configuration that does not enable MPRIS at all, so I wonder why shairport-sync is even trying to connect to dbus...

nicokaiser, 2020-11-22:
Yes, I get the same output when restarting shairport-sync. Does AirPlay work despite this, or does the daemon crash on startup?

giovi321, 2020-11-22:
Shairport sync works :)

nicokaiser, 2021-04-02:
If it works despite the error message, it is fine, however I just published a new version which uses a newer compiled version of shairport-sync (as the Debian version is not maintained anymore). Maybe this solves this warning.

closed 2021-04-02 by nicokaiser