nicokaiser/rpi-audio-receiver #82: No wifi after script installation

Golgomaph, 2020-11-21:
Once i've installed the whole script with all options, or even only the bluetooth-option, the raspberry pi is not able to connect to wifi after. Although it did before.

When using ifconfig it shows no ip for wlan0, i think something with dhcp gets broken.

Anything known to this issue?

nicokaiser, 2020-11-22:
By default, if you did not enable the `ifconfig wlan0 down` commands, nothing is done to the Wi-Fi stack on installation.

- Did you try to connect to it by IP (as the installation script may have changed the hostname?)
- Does it fail to connect at all, or does the connection drop when you connect to it via Bluetooth?
- Do you use an external Bluetooth USB dongle? Can you try to put `dtoverlay=disable-bt` to make sure the internal Bluetooth module is disabled and see if the Wi-Fi behaviour improves?

Golgomaph, 2020-12-12:
Thanks for your answer. 

I noticed that wifi got broken after "sudo apt-get upgrade", it had nothing to do with your script.

closed 2020-12-12 by Golgomaph