nicokaiser/rpi-audio-receiver #84: Annoying voice

3Dscrewer, 2020-12-15:
Sorry for that, I can’t find a solution. I have executed the script and it is working fine.
But if there is no connection, there is repeating a robot voice with something like „to install F3 reader press control on face“ my native language is German and I don’t really understand that voice, even more, if a couple of time repeated, there is only a part of the sentence.
Anyway, any idea to disable that?
Thanks

svenb1234, 2020-12-21:
Did you install with a desktop? If so, you have to attach the screen once, log-in and skip through the RPI first boot wizard. After that the voice should be gone.

3Dscrewer, 2020-12-21:
Sorry, it is a headless System. No Desktop.

nicokaiser, 2020-12-21:
Are you sure you installed Raspberry Pi OS **Lite**? This looks like the `piwiz` tries to follow the initial setup procedure, but as far as I understand (I need to check this) this only happens in the full desktop version, not the "lite" edition...

3Dscrewer, 2020-12-22:
Since my Desktop Version damaged the SDcard and I’m using the Raspi only for PiHole and Airplay with the Hifiberry Digi+ I was lazy and using only that Image. I can’t find an information what Version is used.

https://github.com/RPiList/specials/blob/master/Download.md

I‘m pretty shure, there is no Desktop available.
But uname didn’t say so much

nicokaiser, 2020-12-27:
I don't know SVPihole, so I guess the problem comes from this distribution.
(Also, using the Desktop-version instead of the Lite-version of Raspberry Pi OS may install and activate `piwiz` regardless whether there is a screen attached or not...)

closed 2020-12-27 by nicokaiser

3Dscrewer, 2020-12-28:
To clear the situation. I don’t know why, but the HDMI was pulled half, so I guessed no Desktop.
SVPiHole is using the normal Pi OS with Desktop. After finishing the initial setup at the desktop, I was always using SSH, the voice is gone ....