nicokaiser/rpi-audio-receiver #89: Feature request: Dockerize rpi-audio-receiver

ManuelSchneid3r, 2021-01-19:
Hi Nico, 

awesome work. Thank you. I have a suggestion though. Since the install process is not managed by a package manager the setup can leave your system in a messed up state. I would really appreciate if I could install your work using docker, which guarantees working ootb. 

The rationale is also that this process would make the a2dp feature available for users not using Raspbian. bluezalsa is available on this distro only. Sad because I use ubuntu server on my pi4 because they have (nonexperimental) amd64 support.

If you are not into docker this is pretty much work, but this project fits a absolutely valid usecase. Your range will increase though and maybe, if  you add a donation button, you'll even benefit.

Regards

nicokaiser, 2021-02-11:
Thanks for your suggestion.

I also was thinking about converting this to a Docker container – however there are some problems:

- The shairport-sync Docker image mikebrady/shairport-sync-docker is the only service that is available as container right now
- Getting Bluetooth running in Docker is a real pain and without major implications on the host system it is not possible to get it running (if at all)
- There is no raspotify container
- Adding Docker to the equation would increase the complexity and with its copy-on-write filesystem the startup time would at least double.
- Making a Docker system read-only is not possible at all (at least, without using tons of memory for overlay filesystems).

So having e.g. a Docker Compose file with all the services would not solve the problem of a messed up system. 

The purpose of this script(s) is not providing a clean package that can be installed and removed again, but to (one-way) convert a freshly installed Raspberry Pi into an audio receiver (with the least amout of non-standard system modifications)... it is not about adding audio features to an already installed and heavily customized Raspi OS installation. (Yet, it is more customizable than e.g. HiFiBerry OS, which provides a complete OS image that can only be altered by re-compiling it)

ManuelSchneid3r, 2021-03-12:
> * Getting Bluetooth running in Docker is a real pain and without major implications on the host system it is not possible to get it running (if at all)

You mean the "net":"host" config? I think this is a fair requirement. Usually you dont use BT at all on your pi if you use it as sound sink right?

> * There is no raspotify container

ggoussard/spotifyd works well for me 

> * Adding Docker to the equation would increase the complexity and with its copy-on-write filesystem the startup time would at least double.

Imho this is not really an issue. My pi runs 24/7 anyway. also "works" > "easy setup" > "lightweight" imho.

> * Making a Docker system read-only is not possible at all (at least, without using tons of memory for overlay filesystems).

dont really get this one. with docker system you mean the guest? the docker volumes? or bindmounts? i have a pi4 running with 3 docker services one of them spotify connect. 147MB used from the 2Gb. I thing these are over enginieering thoughts.

but well its your project :D you can do whatever you want. if you dont see the need okay. i just respond because I think the arguments are not _that_ weighty

nicokaiser, 2021-03-13:
> You mean the "net":"host" config? I think this is a fair requirement. Usually you dont use BT at all on your pi if you use it as sound sink right?

Well, part of this is to provide a BT sound sink, and to get this working reliably, some configuration has to be adjusted on the host system. I could not get BlueALSA running in a Docker container (even with Host Network mode), and even if this is possible, it would require quite some modifications on the host system. And, supporting these host system configuration changes for different Docker hosts (such as Raspberry Pi OS, Ubuntu, etc.) is impossible.

> dont really get this one. with docker system you mean the guest? the docker volumes? or bindmounts? i have a pi4 running with 3 docker services one of them spotify connect. 147MB used from the 2Gb. [...]

No, I mean the host: my goal is to provide a system that boots in seconds (~10s for a Raspi 4) and can be switched off without shutdown at any time. Here, I connected a Raspberry Pi Zero to the power outlet of an old audio amplifier, so I can switch on or off the Raspi along with the amplifier. But this means the whole system needs to be read-only (otherwise the filesystem will corrupt at some point) after the Bluetooth clients are configured. This is not possible with Docker.

So my concerns are:

- Bluetooth/BlueALSA is not easy and portable across different host systems with Docker
- Read-only filesystem is not possible with Docker
- Installing Docker is not trivial, if I remember correctly this needs some heavy modifications on a Raspi Zero (this may have improved with Buster)
- Start-up time will be way over 10s with Docker

BUT:

I'd really be interested in a working BlueALSA container at least for ARMv8 Raspberry Pi OS and Ubuntu. Maybe this whole idea of running all services in containers and being able to remove the pain of compiling GitHub sources will work some day...

ManuelSchneid3r, 2021-03-13:
Got it. But ain't it possible to run the entire Bluetooth stack in a container while exposing just the bt device to the container with corresponding privileges? Just an idea I am a total noob considering bluez.

nicokaiser, 2021-03-21:
I'm not sure if this is possible, as the Raspberry Pi Bluetooth stack is quite customized in Raspberry Pi OS, and I have no idea how to build a container with all the customized packages (and how to expose the BT device into it)... I guess this would wildly exceed the complexity of a properly configured, simple, "bare metal" Raspberry Pi OS...

closed 2021-03-21 by nicokaiser