nicokaiser/rpi-audio-receiver #91: How to disable Bluetooth Audio and play System audio from output when needed/desired

Shabinder, 2021-02-03:
This Scripts works great!
Just one thing...
How to go back to play System Audio from Speakers when needed.

Shabinder, 2021-02-03:
Seems Audio works if we change output source for vlc to digital in vlc prefrence but chromium sound still not working!

Seems like Chromium doesnt work well with bluealsa...
Help plz

nicokaiser, 2021-04-02:
I think I did not quite understand what you are trying to do. Would you like to play system audio (like notification sounds) from your local speakers, but VLC sound via Bluetooth on the Raspberry Pi? (This would require some configuration on the client and depends on your setup...)

Shabinder, 2021-04-02:
Never Mind I got it working, thanks for this project

closed 2021-04-02 by Shabinder