nicokaiser/rpi-audio-receiver #92: bt-agent requires manual approval when pairing

stepanmracek, 2021-02-21:
Hi,
thanks for this wonderful set of scripts. I have an issue when I'm trying to connect from the laptop/phone to my raspberry pi. It seems that `bt-agent` ignores config option `--capability=NoInputNoOutput` and always asks for confirmation when I'm pairing some device with raspberry pi:

```
pi@raspberrypi:~ $ journalctl -f
-- Logs begin at Sun 2021-02-21 12:46:30 CET. --
Feb 21 12:58:22 raspberrypi polkitd(authority=local)[1169]: Registered Authentication Agent for unix-process:1164:70657 (system bus name :1.27 [/usr/bin/pkttyagent --notify-fd 5 --fallback], object path /org/freedesktop/PolicyKit1/AuthenticationAgent, locale en_GB.UTF-8)
Feb 21 12:58:25 raspberrypi polkitd(authority=local)[1169]: Operator of unix-process:1164:70657 successfully authenticated as unix-user:pi to gain ONE-SHOT authorization for action org.freedesktop.systemd1.manage-units for system-bus-name::1.29 [systemctl restart bt-agent] (owned by unix-user:pi)
Feb 21 12:58:25 raspberrypi systemd[1]: Stopping Bluetooth Agent...
Feb 21 12:58:25 raspberrypi bt-agent[338]: Authorize this device pairing (yes/no)? unregistering agent...
Feb 21 12:58:25 raspberrypi systemd[1]: bt-agent.service: Succeeded.
Feb 21 12:58:25 raspberrypi systemd[1]: Stopped Bluetooth Agent.
Feb 21 12:58:25 raspberrypi systemd[1]: Started Bluetooth Agent.
Feb 21 12:58:25 raspberrypi polkitd(authority=local)[1169]: Unregistered Authentication Agent for unix-process:1164:70657 (system bus name :1.27, object path /org/freedesktop/PolicyKit1/AuthenticationAgent, locale en_GB.UTF-8) (disconnected from bus)
Feb 21 12:58:25 raspberrypi bt-agent[1177]: Agent registered
Feb 21 12:58:25 raspberrypi bt-agent[1177]: Default agent requested
Feb 21 13:00:33 raspberrypi bt-agent[1177]: Device: BlueZ 5.55 (0C:7A:15:8B:7A:12)
Feb 21 13:00:43 raspberrypi bt-agent[1177]: Authorize this device pairing (yes/no)? Device: BlueZ 5.55 (0C:7A:15:8B:7A:12)
```

If I want to pair I have to run the agent directly like this and type `yes`. After that the devices will be paired:
```
pi@raspberrypi:~ $ bt-agent --capability=NoInputNoOutput
Agent registered
Default agent requested
Device: BlueZ 5.55 (0C:7A:15:8B:7A:12)
Authorize this device pairing (yes/no)? yes
Device: BlueZ 5.55 (0C:7A:15:8B:7A:12) for UUID 0000110d-0000-1000-8000-00805f9b34fb
```

Do you have any idea what I'm doing wrong?

csihama, 2021-02-23:
Hi,
I also have this similar bug:

`pi@raspberrypi:~ $ sudo bt-agent --capability=NoInputNoOutput

Agent registered
Default agent requested
Device: XXXXX (XX:7B:XX:XX:27:88)
Enter passkey: `

Always want to pair with passkey/pin.

nicokaiser, 2021-02-25:
The bt-agent process should never require asking for authorization (this is what "NoInputNoOutput" is about). Are you sure you use an Bluetooth dongle that supports simple pairing mode (SSP, Bluetooth 2.1 or newer)?

The problem with pairing with a PIN is, many devices do not support PIN pairing mechanism anymore, so I removed it as default, see here: https://github.com/nicokaiser/rpi-audio-receiver/issues/48#issuecomment-686666414

closed 2021-02-25 by nicokaiser