nicokaiser/rpi-audio-receiver #95: AirPlay insalled but not shown

SaschaMuehl, 2021-03-07:
Dear Nico, 
thank you for your nice skript. Bluetooth connection is working but the AirPlay Name "MusicPi" isn't shown in my Spotify devices. I also reinstalled the skript with "yes" to all questions. 

I'd send you an log file but I don't know where to find - sry. 

Is the a way to find out why AirPlay isn't shown dispite installation?

Thank you very much

dmtroyer, 2021-03-11:
is this from an iOS 14 device that you are having problems? I am also having problems and have read some anecdotal reports that iOS 14 is forcing airplay 2 (which is unsupported by shairport-sync) .

nicokaiser, 2021-04-02:
shairport-sync works fine even with iOS 14 devices. Can you check if the daemon is running correctly (`systemctl -u shairport-sync`)?
I just pushed a new version which compiles shairport-sync from the sources again, since the Debian package seems to be abandoned again (and outdated)... Maybe you could try to re-install with the newer shairport-sync version and see if this works for you.

closed 2021-04-02 by nicokaiser