nicokaiser/rpi-audio-receiver #99: Bluetooth authentication not working

ManuelSchneid3r, 2021-03-12:
I have no clue if I had to enter a password on the first connect, but I just disconnected because my partner could not connect to the A2DP sink. Now I cant even reconnect myself. I get a message that the device cant connect because the pin or password id incorrect.

mattcd216, 2021-03-19:
Same.  Sure I had it working before...  But now get the pin/passcode error, when trying from multiple devices.
**bluetoothctl show** gives:
`pi@picroft:~ $ bluetoothctl show
Controller E4:5F:01:0C:B7:77 (public)
	Name: picroft
	Alias: picroft
	Class: 0x00040414
	Powered: yes
	Discoverable: yes
	Pairable: yes
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	Modalias: usb:v1D6Bp0246d0532
	Discovering: no
`
Had a little look through bluetoothctl help but couldn't see anything for authenticating.  Any help much appreciated.

kemo87, 2021-03-27:
Same for me.

WiseLord, 2021-03-31:
A also see the same problem. But it appears only in particular situation:

1. When my device (phone) was not previously paired with RPi, the pairing goes OK.
2. When I try connect from the same device more times later, it's also OK.
3. When I delete the RPi from my phone's bluetooth paired list and try to pair again, the pairing FAILS.

It looks like RPi's bluetooth remember old device credentials, and doesn't allow to connect again after being removed from the device.

The solution for me is to delete my device from RPI's bluetooth list. Here is an example of log:

https://gist.github.com/WiseLord/992985b57d0113efbe92ee3ad139911b

Lines 43..44 - is a log for the case when I try to connect again after removing RPi from phone's paired list (fails with a PIN error)
Lines 45..47 - I remove my phone from RPi
Lines 48..81 - is a log for a new attempt, goes OK

WiseLord, 2021-04-02:
I did some additional investigations, and found that switching from custom python script to bt-agent broke the authentification.

https://github.com/nicokaiser/rpi-audio-receiver/commit/a1e6ebd7d6055d002a215de9ae5f306dbf0b8ef7

So, I just switched to the commit before.