nicokaiser/rpi-audio-receiver #101: ALSA VU meter plugin (pivumeter) bug

alphaonex86, 2021-03-26:
update-alternatives: using /usr/bin/automake-1.16 to provide /usr/bin/automake (automake) in auto mode
Processing triggers for man-db (2.8.5-2) ...
Processing triggers for install-info (6.5.0.dfsg.1-4+b1) ...
Processing triggers for libc-bin (2.28-10+rpi1) ...
Cloning into 'pivumeter'...
remote: Enumerating objects: 372, done.
remote: Total 372 (delta 0), reused 0 (delta 0), pack-reused 372
Receiving objects: 100% (372/372), 187.39 KiB | 374.00 KiB/s, done.
Resolving deltas: 100% (176/176), done.
libtoolize: putting auxiliary files in '..'.
libtoolize: linking file '../ltmain.sh'
configure.ac:6: installing './compile'
configure.ac:6: installing './config.guess'
configure.ac:6: installing './config.sub'
configure.ac:3: installing './install-sh'
configure.ac:6: error: required file './ltmain.sh' not found
configure.ac:3: installing './missing'
Makefile.am: installing './INSTALL'
Makefile.am: installing './COPYING' using GNU General Public License v3 file
Makefile.am:     Consider adding the COPYING file to the version control system
Makefile.am:     for your code, to avoid questions about which license your project uses
Makefile.am: installing './depcomp'

nicokaiser, 2021-04-02:
The output `libtoolize: putting auxiliary files in '..'.` seems strange – how did you call the install script?

Weltty, 2021-06-01:
Same issue here. but the log is slightly different. 
I´m using  Pi OS Lite 5.10. 
Do you want to install ALSA VU meter plugin (pivumeter) [y/N] y
Reading package lists... Done
Building dependency tree
Reading state information... Done
autoconf is already the newest version (2.69-11).
automake is already the newest version (1:1.16.1-4).
build-essential is already the newest version (12.6).
git is already the newest version (1:2.20.1-2+deb10u3).
libfftw3-dev is already the newest version (3.3.8-2).
libtool is already the newest version (2.4.6-9).
libasound2-dev is already the newest version (1.1.8-1+rpt1).
wiringpi is already the newest version (2.50).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Cloning into 'pivumeter'...
remote: Enumerating objects: 372, done.
remote: Total 372 (delta 0), reused 0 (delta 0), pack-reused 372
Receiving objects: 100% (372/372), 187.39 KiB | 2.29 MiB/s, done.
Resolving deltas: 100% (176/176), done.
configure.ac:6: installing './compile'
configure.ac:6: installing './config.guess'
configure.ac:6: installing './config.sub'
configure.ac:3: installing './install-sh'
configure.ac:6: error: required file './ltmain.sh' not found
configure.ac:3: installing './missing'
Makefile.am: installing './INSTALL'
Makefile.am: installing './COPYING' using GNU General Public License v3 file
Makefile.am:     Consider adding the COPYING file to the version control system
Makefile.am:     for your code, to avoid questions about which license your project uses
Makefile.am: installing './depcomp'

nicokaiser, 2021-06-15:
Can you try to rename the "install.sh" file in the parent directory (root of this project) to something else? I experienced some very strange behaviour with libtoolize when there is an "install.sh" in a parent directory...

Weltty, 2021-06-15:
I tryed again. this time with a "install.sh" in "/rpi-audio-receiver-main" renamed to "in.sh"  .
But still the same problem:

Cloning into 'pivumeter'...
remote: Enumerating objects: 372, done.
remote: Total 372 (delta 0), reused 0 (delta 0), pack-reused 372
Receiving objects: 100% (372/372), 187.39 KiB | 796.00 KiB/s, done.
Resolving deltas: 100% (176/176), done.
libtoolize: putting auxiliary files in '..'.
libtoolize: linking file '../ltmain.sh'
configure.ac:6: installing './compile'
configure.ac:6: installing './config.guess'
configure.ac:6: installing './config.sub'
configure.ac:3: installing './install-sh'
configure.ac:6: error: required file './ltmain.sh' not found
configure.ac:3: installing './missing'
Makefile.am: installing './INSTALL'
Makefile.am: installing './COPYING' using GNU General Public License v3 file
Makefile.am:     Consider adding the COPYING file to the version control system
Makefile.am:     for your code, to avoid questions about which license your project uses
Makefile.am: installing './depcomp'

nicokaiser, 2021-12-18:
pivumeter is not (really) supported anymore when using PulseAudio, which we must do because Raspberry Pi OS 11 does not include BlueALSA anymore. Pimoroni seems to be working on a "Pi VU Meter" (see [pimoroni/pivumeter-pulseaudio](https://github.com/pimoroni/pivumeter-pulseaudio/)), but it is in a very early stage of development and currently only seems to support the PHAT Beat device.

closed 2021-12-18 by nicokaiser