nicokaiser/rpi-audio-receiver #102: ALAC package not found

hfs, 2021-04-02:
Hi,

just discovered your project (thanks!) and tried to install it for the first time. I said yes to to Shairport out of curiosity and it fails to compile shairport-sync because the ALAC library is missing.

`install-shairport.sh` calls configure with `--with-apple-alac`, but alac is not installed.

```
configure.ac:21: installing './compile'
configure.ac:6: installing './install-sh'
configure.ac:6: installing './missing'
Makefile.am: installing './INSTALL'
Makefile.am: installing './depcomp'
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking for ranlib... ranlib
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of gcc... gcc3
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking dependency style of g++... gcc3
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for clock_gettime in -lrt... yes
checking for pthread_create in -lpthread... yes
checking for exp in -lm... yes
>>Including libpopt
checking for POPT... yes
>>Including the dummy audio back end
checking for LIBCONFIG... yes
>>Including the Apple ALAC Decoder
checking for ALAC... no
configure: error: Package requirements (alac) were not met:

No package 'alac' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables ALAC_CFLAGS
and ALAC_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.
```

Should I get ALAC from https://github.com/TimothyGu/alac ?

it-monkey, 2021-04-06:
Thanks @TNTLarsn  & @hfs  installing from https://github.com/TimothyGu/alac worked for me as well. Wrote an article on how to install it completely and added this to the troubleshooting section -> https://medium.com/codex/building-a-raspberry-pi-audio-receiver-6e579b7ce003

nicokaiser, 2021-04-08:
Nice catch, I added the ALAC installation to the shairport-sync script.

commit e579b0e4 (2021-04-08): fix: add ALAC installation to sharport-sync (fixes #102)