nicokaiser/rpi-audio-receiver #104: Raspotify wrong config file

Arcadia197, 2021-04-07:
I didn't have a deeper look if this is created by the script or something from raspotify itself, but the file under _/etc/default/raspotify_ has some wrong configurations after setup for me.

In particular: the line with VOLUME_ARGS is causing issues. I had to change it manually to:
_VOLUME_ARGS="--enable-volume-normalisation --volume-ctrl linear --initial-volume=100"_
and to restart the service with
_sudo systemctl restart raspotify_
to get the device to show up in spotify successfully

Tested on my RPi 0 (two seperate ones) and RPi 4

nicokaiser, 2021-04-08:
Which issues is the line creating? Apart from "--enable-volume-normalisation" the generated config file is the same as the VOLUME_ARGS you posted... Or did I miss somthing?

Arcadia197, 2021-04-08:
I'm sorry I didn't include further information about the error yet. I will setup another RPI 0 soon and will then provide the full created config file after the log and the syslog error from raspotify.