nicokaiser/rpi-audio-receiver #105: shairport-sync.service: libalac.so.0 not found

TwizzyDizzy, 2021-04-14:
Hi @nicokaiser 

just used this repo to set up Bluetooth / Airplay on my `Raspberry Pi 3 B` on `Raspberry Pi OS Lite Buster` at commit https://github.com/nicokaiser/rpi-audio-receiver/commit/26d6a31fbf3732825a5969cc1b60202c985e3756 of this repo.

One thing that did not work out of the box: Airplay.

It seems as though the compiled version of `libalac` is put into `/usr/lib` ... the `LD_LIBRARY_PATH` does not contain this path though, so we have to set in the `shairport-sync.service`:

```
systemctl edit shairport-sync.service

[put the line with 'Environment=' in there]

# /etc/systemd/system/shairport-sync.service.d/override.conf
[Service]
# Avahi daemon needs some time until fully ready
ExecStartPre=/bin/sleep 3
Environment=LD_LIBRARY_PATH=/usr/local/lib
```

Cheers
Thomas

TwizzyDizzy, 2021-04-14:
Submitted a PR to address this, see comments there: https://github.com/nicokaiser/rpi-audio-receiver/pull/106

nicokaiser, 2021-04-24:
This is strange, since shairport-sync is also compiled from source (thus, installed to `/usr/local/bin/shairport-sync`), so it should find its libraries in `/usr/local/lib`.

I'll try to install a fresh version of the script to see what's wrong and if the LD_LIBRARY_PATH is really needed...

nicokaiser, 2021-04-24:
@TwizzyDizzy Can you check if this works: the current install script I just pushed runs `ldconfig` after installing `alac`. This should fix the issue without needing to provide a `LD_LIBRARY_PATH`.

closed 2021-04-24 by nicokaiser

TwizzyDizzy, 2021-04-24:
Yeah, this solution is to be preferred over the workaround in the systemd unit which is what I meant with

> Might also be solved by `configure`ing different LDFLAGS here: e579b0e.

in the PR.

I'll give it a spin with a spare SD and your pushed version, probably tomorrow, and report back.

Cheers
Thomas

TwizzyDizzy, 2021-04-24:
Ah.. did it now so we can finish this: can confirm, your `ldconfig` does the trick.

This works due to:

```
root@livingroom:~# cat /etc/ld.so.conf.d/libc.conf 
# libc default configuration
/usr/local/lib
```

The output of `ldd` now shows: `libalac.so.0` is found:

```
root@raspberrypi:/opt/rpi-audio-receiver# ldd /usr/local/bin/shairport-sync
	linux-vdso.so.1 (0x7ee0b000)
	/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so => /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so (0x76f23000)
	libmosquitto.so.1 => /lib/arm-linux-gnueabihf/libmosquitto.so.1 (0x76f06000)
	libgio-2.0.so.0 => /lib/arm-linux-gnueabihf/libgio-2.0.so.0 (0x76d8a000)
	libgobject-2.0.so.0 => /lib/arm-linux-gnueabihf/libgobject-2.0.so.0 (0x76d2d000)
	libglib-2.0.so.0 => /lib/arm-linux-gnueabihf/libglib-2.0.so.0 (0x76c1d000)
	libsndfile.so.1 => /lib/arm-linux-gnueabihf/libsndfile.so.1 (0x76ba1000)
	libasound.so.2 => /lib/arm-linux-gnueabihf/libasound.so.2 (0x76ac0000)
	libavahi-common.so.3 => /lib/arm-linux-gnueabihf/libavahi-common.so.3 (0x76aa5000)
	libavahi-client.so.3 => /lib/arm-linux-gnueabihf/libavahi-client.so.3 (0x76a86000)
	libsoxr.so.0 => /lib/arm-linux-gnueabihf/libsoxr.so.0 (0x76a24000)
	libssl.so.1.1 => /lib/arm-linux-gnueabihf/libssl.so.1.1 (0x769a5000)
	libcrypto.so.1.1 => /lib/arm-linux-gnueabihf/libcrypto.so.1.1 (0x7678b000)
	libalac.so.0 => /usr/local/lib/libalac.so.0 (0x76772000)
	libconfig.so.9 => /lib/arm-linux-gnueabihf/libconfig.so.9 (0x76758000)
	libpopt.so.0 => /lib/arm-linux-gnueabihf/libpopt.so.0 (0x7673d000)
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x76713000)
	librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0x766fc000)
	libstdc++.so.6 => /lib/arm-linux-gnueabihf/libstdc++.so.6 (0x765b5000)
	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x76533000)
	libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0x76506000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x763b8000)
	/lib/ld-linux-armhf.so.3 (0x76f38000)
	libgmodule-2.0.so.0 => /lib/arm-linux-gnueabihf/libgmodule-2.0.so.0 (0x763a4000)
	libz.so.1 => /lib/arm-linux-gnueabihf/libz.so.1 (0x76379000)
	libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0x76366000)
	libmount.so.1 => /lib/arm-linux-gnueabihf/libmount.so.1 (0x76305000)
	libselinux.so.1 => /lib/arm-linux-gnueabihf/libselinux.so.1 (0x762d3000)
	libresolv.so.2 => /lib/arm-linux-gnueabihf/libresolv.so.2 (0x762af000)
	libffi.so.6 => /lib/arm-linux-gnueabihf/libffi.so.6 (0x76297000)
	libpcre.so.3 => /lib/arm-linux-gnueabihf/libpcre.so.3 (0x76220000)
	libFLAC.so.8 => /lib/arm-linux-gnueabihf/libFLAC.so.8 (0x761c0000)
	libogg.so.0 => /lib/arm-linux-gnueabihf/libogg.so.0 (0x761ab000)
	libvorbis.so.0 => /lib/arm-linux-gnueabihf/libvorbis.so.0 (0x76174000)
	libvorbisenc.so.2 => /lib/arm-linux-gnueabihf/libvorbisenc.so.2 (0x760e2000)
	libdbus-1.so.3 => /lib/arm-linux-gnueabihf/libdbus-1.so.3 (0x76091000)
	libgomp.so.1 => /lib/arm-linux-gnueabihf/libgomp.so.1 (0x76059000)
	libblkid.so.1 => /lib/arm-linux-gnueabihf/libblkid.so.1 (0x76001000)
	libsystemd.so.0 => /lib/arm-linux-gnueabihf/libsystemd.so.0 (0x75f64000)
	libuuid.so.1 => /lib/arm-linux-gnueabihf/libuuid.so.1 (0x75f4d000)
	liblzma.so.5 => /lib/arm-linux-gnueabihf/liblzma.so.5 (0x75f1c000)
	liblz4.so.1 => /lib/arm-linux-gnueabihf/liblz4.so.1 (0x75ef0000)
	libgcrypt.so.20 => /lib/arm-linux-gnueabihf/libgcrypt.so.20 (0x75e16000)
	libgpg-error.so.0 => /lib/arm-linux-gnueabihf/libgpg-error.so.0 (0x75dea000)
```

Cheers
Thomas

commit 768b4d8b (2021-04-24): fix: run ldconfig after installing alac (fixes #105)