nicokaiser/rpi-audio-receiver #110: Bluetooth requires authentication with iOS 14

udomoody, 2021-06-10:
I tested with my one iPhone on iOS 12, my other iPhone on iOS 14 and my laptop running win10.
Both the iOS 12 iPhone and my laptop connect over bluetooth without any problem but on my iOS 14 iPhone I always got a `Pairing Unsuccessful` message. 
I checked the logs on my pi and saw it required a manual authentication. To do this I need to run `bt-agent` on my pi, then try to connect with my iPhone so it shows a pin code that I have to confirm with `yes `on the pi. 

Does anyone know why this happens for iOS 14 devices apparently?

lostmyshape, 2021-06-10:
iOS 14.6 works fine for me.  However, I had the same issue with my Android phone.  Every other device in my house connected normally, without manual authentication, but that one wouldn't.  I have no idea why.  After authenticating manually once, it would connect without needing to authenticate again.

I ended up, re-installing RPi OS lite and the rpi-audio-reciever script again and it went back to normal.

pvanthony, 2021-08-17:
Had a similar issue. Checkout this link.
https://github.com/JasonLG1979/PiZero-Bluetooth-Audio-Receiver/issues/2