nicokaiser/rpi-audio-receiver #113: Please clarify if Raspberry Pi OS (Legacy) can still be used

HorstBaerbel, 2021-12-04:
The RPi foundation released a [backwards-compatible version of Raspbian / Raspberry Pi OS called "Legacy"](https://www.raspberrypi.com/software/operating-systems/), base of Debian "Buster". Can this be used together with rpi-audio-receiver (bluealsa package etc.)? Would be good to have that info in the README.

nicokaiser, 2021-12-05:
There is a `debian-10` branch that still runs on Raspberry Pi OS 10 (the "Legacy" version). 

Newer updates of this repository will only work with the latest version (Raspberry Pi OS 11), with PulseAudio instead of BlueALSA. (This may or may not work on slower devices like Raspberry Pi 1 or Pi Zero)

closed 2021-12-05 by nicokaiser

HorstBaerbel, 2021-12-05:
Thanks for the info. A hint in the README couldn't hurt...