nicokaiser/rpi-audio-receiver #119: Bluetooth authentification does not work after completely disconnect and reconnect

manuelambaum, 2022-01-08:
Hi, i have the same issue as discussed in the previous post: https://github.com/nicokaiser/rpi-audio-receiver/issues/99
If i completely disconnect the bluetooth connection from my phone/laptop to the raspberry pi, the is the error with wrong pin or password. I think i installed the newest version of this library, but it semms that it does not work. Can anyone help me? thanks

nicokaiser, 2022-01-08:
This is strange, you should never need to enter a PIN or password. I’ll have a look at it next week.

arxeiss, 2022-04-14:
Maybe not related, but I also have to enter PIN. But not with new Bluetooth USB dongle.
This is what I did from the beginning:

closed 2023-08-13 by nicokaiser