nicokaiser/rpi-audio-receiver #121: Not able to set volume of Audio speaker with Windows Computer

manuelambaum, 2022-01-11:
Hi, if i turn on the bluetooth, on my Windows Computer and connect with the Bluetooth of the Raspberry, i get a strange issue:
I watched a Video on youtube and if I adjust the Video-Volume directly on youtube, everything works fine. But if i adjust the Volume on the bottom right on my Windows Computer(which is generally the Volume of all Audio sources on the Computer), the output volume on my raspberry pi does not change. Same problem with spotify. If i adjust the volume directly over the slider in the Spotify app, everything works, but also in this case, windows general volume does not change the volume on the raspberry. I tested it on my android phone, but it seems that these problems only exist on my Computer and my Android Phone is not affected of this issue

bhborba, 2022-01-16:
I have the exact same issue with my Apple TV over bluetooth

thelonius-klute, 2022-01-21:
I had the same issue with iPhones and found this solution worked:
https://github.com/hadess/CHIP-bluetooth-speaker/issues/8#issuecomment-264014153

bhborba, 2022-01-23:
Worked for me as well. Thanks!

nicokaiser, 2022-02-04:
Which of the solutions worked? Upgrading BlueZ to testing or setting the `--plugin a2dp` parameter? (The latter completely broke Bluetooth for my setup).

thelonius-klute, 2022-02-07:
I didn't upgrade to the test version, so assuming I was already on the latest version of BlueZ then setting the parameter must have fixed it.

mousemat86, 2023-07-03:
is there a new way to solve this issue? I tried the above but it completley broke my setup. 

I can see the bluetooth service installed with the script already includes `--plugin=a2dp` but the problem persists. Is it due to an outdated version of pulseaudio-module-bluetooth? 

Thanks!

nicokaiser, 2023-07-20:
I'm sorry, since I am not using Bluetooth anymore, I cannot test this. Bluetooth in Linux is a horrible moving target, so everything changes from Debian (especially Raspbian) release to the next...

closed 2023-08-13 by nicokaiser