nicokaiser/rpi-audio-receiver #123: Menu bar is gone

dasdirksche, 2022-01-22:
Hi there, after I run this script I got the new user "[benutzer]" and the most annoying thing:  my dock (Anwedungsleiste) is gone. If I connect via VNC I got no graphical possibility to interact with the system.
RaspberryPiOs bullseye, Raspberry Pi 3+

nicokaiser, 2022-01-22:
Did you follow the instructions and use only the "Lite" version (without graphical interface) for your device?
This script is meant to convert a freshly installed (!) Raspberry Pi device with Raspberry Pi OS Lite (!) into an audio receiver. It is not meant to modify an existing system with graphical UI and so on.

closed 2022-01-22 by nicokaiser

dasdirksche, 2022-01-22:
Okay. Got it. I used it on a system with running Pi-hole, ejabberd Server etc. Everything is working fine by the way.