nicokaiser/rpi-audio-receiver #125: Uninstall

Photon99984, 2022-02-03:
What is the easiest way to uninstall or at least disable the bluetooth audio-receiver permanently?

nicokaiser, 2022-02-03:
Short answer: flash the SD card.

Long answer: this script is meant to be used on a clean Raspberry Pi OS Lite installation. It is not meant to be run on a device that does other things. So, it permanently modifies the installation, it is not meant to be uninstalled at all.

It you really need to disable parts of it, you need to look at the scripts and try to reverse their behaviour (although some files are overwritten).

closed 2022-02-03 by nicokaiser

Photon99984, 2022-02-04:
Thanks for the quick answer and for making this awesome project. No problem for me, but I think you should mention that in the readme.