nicokaiser/rpi-audio-receiver #131: Music stops playing

Zebsi235, 2022-03-15:
Hi, so yesterday I setup my Raspberry to use it as a BT receiver for my car. When I tried it at home everything worked fine, but when I pluged it in inside the car the Music sometimes just stopped playing. I dont really know if thats a problem of this repo or if its just unsteady Voltage that causes this to happen. The wierd thing is whenever it stopped playing, I could just skip to the next Song and it would work again for 2 to 3 Minutes. Any ideas on how to fix that ? 

Thanks a lot

Mikka123T, 2022-03-19:
Hi, I had similar problem but only when running mopidy. I have had to change /etc/pulse/client.conf to
#default-server = /run/pulse/native
#autospawn = no

nicokaiser, 2022-05-19:
Sorry, I cannot reproduce this. Could you test if this is a voltage problem or related to mopidy or something?

closed 2022-05-19 by nicokaiser