nicokaiser/rpi-audio-receiver #142: Question:  Bluetooth Audio to AirPlay device

KuzmaTolyanuch, 2022-08-15:
Hi @nicokaiser,
Thank you for your effort on this project, it is very useful. 
I did install it on Orange Pi Zero 2 and have working Bluetooth Audio, AirPlay 2 (separate shairport-sync in docker),
And now I'm looking for a way to add another functionality - routing Bluetooth Audio to Another AirPlay device.
The purpose is to get ability to stream audio to HomePod from Android device using Bluetooth. 
Could you please advise, is it possible to do such thing taking this solution as basis?

nicokaiser, 2022-08-15:
Sorry, I cannot help here. The scope of this project is getting basic audio receiver capabilities on a Raspberry Pi using Raspbian.

I am not sure if this is possible at all, since Streaming to HomePod requires an AirPlay 2 sender (not receiver)…

closed 2022-08-15 by nicokaiser

KuzmaTolyanuch, 2022-08-16:
Make sense. Thanks for response anyway. 
I've found a way to do it BTW. There is a raop plugin for pulseaudio - https://hfujita.github.io/pulseaudio-raop2 , which make it possible, seems to be tricky but I will figure it out.