nicokaiser/rpi-audio-receiver #147: Bluetooth udev script regex breaking logic

elderlabs, 2023-05-13:
This refers to `/usr/local/bin/bluetooth-udev`.

With the regex device name check `if [[ ! $NAME =~ ^\"([0-9A-F]{2}[:-]){5}([0-9A-F]{2})\"$ ]]; then exit 0; fi` enabled, devices that connect to Debian 11 do not match a mac address naming scheme -- or at the very least, a Windows device doesn't. The device connects with its hostname.

This breaks the logic flow the script intends as it doesn't disable discovery upon connection, doesn't disable wifi for those that have issues with it (wifi doesn't cause an issue on my end), and doesn't play a device connection sound upon add/remove.

My solution for this was to comment out the name check, which allowed the script to run as intended. Given we're filtering for input devices. I suppose that'd possibly catch other input devices, like a keyboard/mouse, but as this is an audio sink, other devices will likely never be connected to it.

quite, 2024-01-16:
Hm, yes that check is a bit odd. None of my phones or computers that connected have a `NAME` that was hexadecimal, they all had their set device name (sometimes with a parenthesis after).

@elderlabs a device connection sound? I never heard one, and didn't see anything in the scripts, what did I miss...

rriemann, 2024-04-24:
I also doubt my udev triggers the script. I got:

~~~
udevadm monitor --environment --udev
monitor will print the received events for:
UDEV - the event which udev sends out after rule processing

UDEV  [958.788558] remove   /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.0/bluetooth/hci0/hci0:70 (bluetooth)
ACTION=remove
DEVPATH=/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.0/bluetooth/hci0/hci0:70
SUBSYSTEM=bluetooth
DEVTYPE=link
SEQNUM=2338
USEC_INITIALIZED=593747535
SYSTEMD_ALIAS=/sys/subsystem/bluetooth/devices/hci0:70
SYSTEMD_WANTS=bluetooth.target
SYSTEMD_USER_WANTS=bluetooth.target
TAGS=:systemd:
CURRENT_TAGS=:systemd:

                                   
    

UDEV  [1039.414210] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.0/bluetooth/hci0/hci0:70 (bluetooth)
ACTION=add
DEVPATH=/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.0/bluetooth/hci0/hci0:70
SUBSYSTEM=bluetooth
DEVTYPE=link
SEQNUM=2339
USEC_INITIALIZED=1039412512
SYSTEMD_ALIAS=/sys/subsystem/bluetooth/devices/hci0:70
SYSTEMD_WANTS=bluetooth.target
SYSTEMD_USER_WANTS=bluetooth.target
TAGS=:systemd:
CURRENT_TAGS=:systemd:
~~~

I would expect to hear a sound, but I do not:

~~~
cat /usr/local/bin/bluetooth-udev
#!/bin/bash
# if [[ ! $NAME =~ ^\"([0-9A-F]{2}[:-]){5}([0-9A-F]{2})\"$ ]]; then exit 0; fi

action=$(expr "$ACTION" : "\([a-zA-Z]\+\).*")

if [ "$action" = "add" ]; then
    aplay -q /usr/local/share/sounds/__custom/device-added-LN.wav
    bluetoothctl discoverable off
    # disconnect wifi to prevent dropouts
    #ifconfig wlan0 down &
fi

if [ "$action" = "remove" ]; then
    aplay -q /usr/local/share/sounds/__custom/device-removed-LN.wav
    # reenable wifi
    #ifconfig wlan0 up &
    bluetoothctl discoverable on
fi

~~~

Kirs-on, 2026-11-03:
Hi! I found a walkaround for regex check at the begging of the script. In my case NAME wasn't a MAC address of a device but literal name, so I'm checking PRODUCT variable to see if anything connected/ disconnected via bluetooth, if the value starts with 5/ - I'm assuming that happened. 

so at the begging of   ` /usr/local/bin/bluetooth-udev `

I use
~~~
if [[ ! $PRODUCT =~ ^5/ ]]; then exit 0; fi
~~~

instead of 

~~~
if [[ ! $NAME =~ ^\"([0-9A-F]{2}[:-]){5}([0-9A-F]{2})\"$ ]]; then exit 0; fi
~~~

works or me